Freescale/meta-freescale #11: IMAGE_BOOT_FILES broken after recent kernel-devicetree.bbclass changes

jonathanhaigh-arm, 2018-07-12:
**Description**
The recent openembedded-core commit [1860d9d3c62e2e94cd68a809385873ffd8270b6d](https://github.com/openembedded/openembedded-core/commit/1860d9d3c62e2e94cd68a809385873ffd8270b6d) changed the names of DTB files in the deploy directory, but meta-freescale's IMAGE_BOOT_FILES content hasn't been updated to reflect this. 

The DTB related contents of IMAGE_BOOT_FILES is generated using the make_dtb_boot_files function in conf/machine/include/utilities.inc which expects the file names of the DTBs in the deploy directory to be prefixed with ${KERNEL_IMAGETYPE}, but they no longer are.

**Steps to reproduce the issue:**
1. Build an image with MACHINE=imx7s-warp and "wic.gz" in IMAGE_FSTYPES. 

**Describe the results you received:**
```
ERROR: my-image-1.0-r0 do_image_wic: Function failed: do_image_wic (log file is located at /data/w7/build/tmp/work/imx7s_warp-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.5799)
ERROR: Logfile of failure stored in: /data/w7/build/tmp/work/imx7s_warp-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.5799
Log data follows:
```
...
```
| DEBUG: Executing shell function do_image_wic
| INFO: Creating image(s)...
| 
| ERROR: _exec_cmd: install -m 0644 -D /data/w7/build/tmp/deploy/images/imx7s-warp/zImage-imx7s-warp.dtb /data/w7/build/tmp/work/imx7s_warp-oe-linux-gnueabi/my-image/1.0-r0/deploy-my-image-image-complete/my-image-imx7s-warp-20180711143641/tmp.wic.sqhyghzx/boot.2/imx7s-warp.dtb returned '1' instead of 0
| output: install: cannot stat '/data/w7/build/tmp/deploy/images/imx7s-warp/zImage-imx7s-warp.dtb': No such file or directory
| 
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_image_wic (log file is located at /data/w7/build/tmp/work/imx7s_warp-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.5799)
```

**Describe the results you expected:**
No build errors.

**Additional details (revisions used, host distro, etc.):**
meta-freescale version: ce51c26a125495728b9d5485afa4d540bf0f405b
oe-core version: f9324af88a99eca28b160fa31aa4516fd397e44b

otavio, 2018-07-12:
It has been fixed with 791a4e76

closed 2018-07-12 by otavio

jonathanhaigh-arm, 2018-07-12:
Great, thanks Otavio.