Freescale/meta-freescale #18: build(core-image-sato): conflict with remove_imxgpu2d and matchbox

cpboyd, 2018-10-11:
I ran into an issue on the rocko release and noticed this thread for morty:
https://lists.yoctoproject.org/pipermail/poky/2017-June/011007.html

Simply commenting out these lines from `recipes-graphics/gtk+/gtk+3_%.bbappend` allowed the build to continue:
```bb
# PACKAGECONFIG_remove_imxgpu2d = " \
#     ${@bb.utils.contains("DISTRO_FEATURES", "wayland", "x11", "", d)} \
# "
```

That is still present in the current master & sumo branches, so presumably the build issue is still present in the pending sumo release.

otavio, 2018-10-15:
What is your `DISTRO_FEATURES` value?

cpboyd, 2018-10-15:
I used `DISTRO` `fslc-xwayland` which has:
```
# Remove conflicting backends
DISTRO_FEATURES_remove = "directfb"
DISTRO_FEATURES_append = " x11 wayland pam systemd"
```

otavio, 2018-10-15:
Humm, this seems to be the culprit. Could you test using x11?

cpboyd, 2018-10-16:
Using `fslc-x11` instead seems to eliminate the build issue.

closed 2018-10-26 by cpboyd