Freescale/meta-freescale #27: gtk+3_%.bbappend causes build failure

cdgarren, 2018-12-03:
The bbappend for gtk+3 is causing chromium-x11 to fail. It relies on the gtk x11 headers to be present, but this removes x11 from the distro features, and causes the headers to not be generated. Commenting out the bbappend contents allows the build to succeed.

otavio, 2018-12-03:
Which machine and distro did you use?

cdgarren, 2018-12-03:
Machine is var-som-mx6, and distro is poky. I was using the imx7-var-som and was able to build successfully.

otavio, 2018-12-03:
Which branch?

cdgarren, 2018-12-03:
I'm on the sumo branch.

thochstein, 2018-12-03:
Originally the bbappend disabled Wayland backend if x11 was in the distro. Since we now use Wayland backend for i.MX GPU if both wayland and x11 are in the distro, the disablement was reversed. I don't know  the ramifications of dropping the x11 disablement. I'll try to ask Prabhu. Is it possible to use a Wayland version of chromium instead?

cdgarren, 2018-12-04:
My schedule doesn't allow time to work on using chromium-wayland. One suggestion I have is to possibly warn or error if both distro features are enabled, instead of silently removing one.

otavio, 2019-01-29:
Any news on this side? otherwise I will close this.

otavio, 2019-04-05:
Close due inactiveness.

closed 2019-04-05 by otavio