Freescale/meta-freescale #35: Receipes that depend on libusb are broken

deinok, 2019-01-14:
They should depend on "libusb1"

angolini, 2019-01-14:
Would you mind to prepare a PR?

deinok, 2019-01-14:
@angolini sure, just let me run a `bitbake world` 
Also, a normal PR to github.com repo is okey (Not to yoctoproject.org or via email)?

angolini, 2019-01-14:
From meta-freescale, I was able to stop only 2 recipes with DEPENDS = "libusb"

```
recipes-bsp/mxsldr/mxsldr_git.bb:5:DEPENDS = "libusb"
recipes-devtools/imx-usb-loader/imx-usb-loader_git.bb:6:DEPENDS = "libusb"
```

Some other recipe already depends on libusb1

If you need to update some recipe from yocto or openembedded, than it's better to send the patch to the proper mailing list.

If it's meta-freescale related recipe, you can choose what you prefer. Some PR are being accepted without being sent to the community, and all the patches from the community is merged.

I'm not quite sure I fully understand your last comment, so please apologize if I'm being verbose.

deinok, 2019-01-14:
Yep, these two are the only one that cause compilation issues.
A `bitbake world` runs well.

> I'm not quite sure I fully understand your last comment, so please apologize if I'm being verbose.

Don't worry, its just that some repositories ask contributors to follow a diferent way to send patches

PR #37 (2019-01-15): Move libusb receipes to use libusb1
Closes #35
closed 2019-01-15 by otavio