Freescale/meta-freescale #39: imx-boot_0.2.bb do_compile Failure

NguyenTrongThinh, 2019-01-17:
**ISSUE**
| NOTE: Copy ddr_firmware: lpddr4_pmu_train_2d_dmem.bin from /home/user1/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/deploy/images/imx8mqevk -> /home/user1/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/work/imx8mqevk-fsl-linux/imx-boot/0.2-r0/git/iMX8M
| **cp: cannot stat '/home/user1/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/deploy/images/imx8mqevk/imx-boot-tools/u-boot-nodtb.bin': No such file or directory**
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_compile (log file is located at /home/user1/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/work/imx8mqevk-fsl-linux/imx-boot/0.2-r0/temp/log.do_compile.4266)
**ERROR: Task (/home/user1/DATA2/SeiWhale/Yocto/sources/meta-freescale/recipes-bsp/imx-mkimage/imx-boot_0.2.bb:do_compile) failed with exit code '1'**
NOTE: Tasks Summary: Attempted 3585 tasks of which 1417 didn't need to be rerun and 1 failed.
NOTE: Writing buildhistory

**_=> recipes imx-boot_0.2.bb failed at do_compile because cannot find the file u-boot-nodtb.bin_**

**My Environment** 

1. Ubuntu 18.04
2. meta-freescale Branch Thud
3. Meta-freescale-distro
4. some metadata need for my distro.

Steps to reproduce:

1. Select distro fslc-wayland (contains in meta-freescale-distro)
2. bitbake core-image-weston

angolini, 2019-01-17:
let me try it on my side.

angolini, 2019-01-17:
Can you, please try https://github.com/angolini/meta-freescale/tree/fix_u-boot_nodtb

So I can create the PR and merge

NguyenTrongThinh, 2019-01-18:
@angolini, I have just tried your meta-freescale branch fix_u-boot_nodtb. I confirmed that no error occurs while building core-image-weston. Below are some lines in log:

**meta-freescale       = "fix_u-boot_nodtb:5a7d790a0b9a49f1446acd44905feaac5aafa115"**
meta-freescale-distro = "thud:771fb6d4c0c9530083fcc8c5452270bb9b3915ba"

Initialising tasks: 100% |#######################################| Time: 0:00:02
Sstate summary: Wanted 465 Found 25 Missed 440 Current 995 (5% match, 69% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
NOTE: Tasks Summary: Attempted 4277 tasks of which 3542 didn't need to be rerun and all succeeded.
NOTE: Writing buildhistory

Summary: There was 1 WARNING message shown.

Please Merge this fix to meta-freescale!

otavio, 2019-01-29:
I merged #41. Closing it.

closed 2019-01-29 by otavio