Freescale/meta-freescale #53: u-boot-imx-mfgtool compiled with error

NguyenTrongThinh, 2019-03-06:
**My environment** 
meta-freescale: thud
meta-openembedded: thud
poky: thud
Ubuntu 18.04
**ISSUE:**
| DEBUG: Python function extend_recipe_sysroot finished
| DEBUG: Executing shell function do_install
**| install: cannot create regular file '/home/thinhnguyen/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/work/seiwhale-seiwhale-linux/u-boot-imx-mfgtool/2017.03-r0/image/boot/spl/u-boot-spl.bin-seiwhale-mfgtool-2017.03-r0-mfgtool-2017.03-r0': No such file or directory**
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_install (log file is located at /home/thinhnguyen/DATA2/SeiWhale/Yocto/build-seiwhale/tmp/work/seiwhale-seiwhale-linux/u-boot-imx-mfgtool/2017.03-r0/temp/log.do_install.12427)
ERROR: Task (/home/thinhnguyen/DATA2/SeiWhale/Yocto/sources/meta-freescale/recipes-bsp/u-boot/**u-boot-imx-mfgtool_2017.03.bb**:do_install) failed with exit code '1'
**Reproduce Steps:**
1. Select Machine: imx8mqevk
2. Set UBOOT_CONFIG = "mfgtool"
3. bitbake u-boot-imx-mfgtool

angolini, 2019-06-13:
Which branch?

Would you mind testing with master?

NguyenTrongThinh, 2019-06-19:
Branch: Thud
I have not tested with master. will try

bacol89, 2019-09-17:
I'm having the same problem. I tried on warrior and master branch.

otavio, 2021-04-10:
@thochstein can you check this?

thochstein, 2021-04-10:
2019?

thochstein, 2021-04-10:
> 2019?

... and apparently still broken...

otavio, 2021-04-10:
It seems people are not using mfgtool very often...

angolini, 2021-08-06:
let me try to build it on master today

thochstein, 2021-08-07:
Looks like this is the same error fixed by #825.

otavio, 2021-08-07:
I'll consider this as fixed then. Closing.

closed 2021-08-07 by otavio