Freescale/meta-freescale #58: Adding a white-space into each (append|+=) variable

IniterWorker, 2019-03-18:
# Summary

Many append variables in meta-freescale haven't a white a the string start. According to the official documentation, all append variable require a white-space to prevent the #57 problem.

## Official Documentation

### [3.1.8. Appending and Prepending (Overrid#e Style Syntax)](https://www.yoctoproject.org/docs/1.6/bitbake-user-manual/bitbake-user-manual.html)
B_append = " additional data"

## Question

Do you want a pull request on `sumo` branch with all replaced appended variables?

otavio, 2019-03-19:
Sure but please check the new branches as well.

IniterWorker, 2019-03-19:
@otavio something like that or you have any other preference.

IniterWorker, 2019-03-19:
I forget something in the regex. I will modify my PRs in few hours @otavio

otavio, 2019-04-05:
All PR were applied. Closing it now.

closed 2019-04-05 by otavio