Freescale/meta-freescale #66: Wrong when populate SDK

ruibailin, 2019-03-20:
I want to setup a standalone compiling environment from yocto project, but meet with the following wrong: 

$ DISTRO=fsl-imx-wayland MACHINE=imx8mqevk bitbake core-image-base –c populate_sdk                                                                                                                                                                                                                 Loading cache: 100% |############################################| Time: 0:00:00
Loaded 3299 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:02
Parsing of 2450 .bb files complete (2448 cached, 2 parsed). 3301 targets, 439 skipped, 9 masked, 0 errors.
ERROR: Nothing PROVIDES '–c'

Summary: There was 1 ERROR message shown, returning a non-zero exit code.

Can anyone help me for that?

Thanks in advance.

otavio, 2019-03-20:
@thochstein would be in a better position to reply as this seems to be using the NXP SDK.

thochstein, 2019-03-26:
Where did you get that command line from? You shouldn't specify the DISTRO in a bitbake since the DISTRO should not change and should already be set in your build folder config. It's okay to set MACHINE, though it is un-necessary if you already set up your build folder with that MACHINE.

The error message looks to be specifically due to a typo. You have an 'en dash' on the command line so bitbake is looking for a package with the name <en-dash>c.

otavio, 2019-04-05:
Seems like it is indeed a typo from the user. I am hence closing this issue.

closed 2019-04-05 by otavio