Freescale/meta-freescale #93: imx-gpu-viv issue with xserver-xorg

cpboyd, 2019-05-15:
Is there a reason this hasn't been updated to version `6.2.4.p4.0` as on https://source.codeaurora.org/external/imx/meta-fsl-bsp-release/tree/imx/meta-bsp/recipes-graphics/imx-gpu-viv?h=sumo-4.14.98-2.0.0_ga ?

I still get a vague seg fault if unless I rollback `xserver-xorg` to `1.19.6` like in Variscite's BSP: https://github.com/varigit/meta-variscite-fslc/commit/5e18d511626c86dd69cf206e2d66a3c34a31c45c

```
(EE) 
(EE) Backtrace:
(EE) 
(EE) Segmentation fault at address 0x4
(EE) 
Fatal server error:
(EE) Caught signal 11 (Segmentation fault). Server aborting
(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
```

angolini, 2019-05-15:
Do you mean #90 ?

cpboyd, 2019-05-15:
@angolini I do.  Thanks!
 I'd searched prior issues, but should've paid more attention to pending PRs. :laughing:

angolini, 2019-05-15:
If you test, please let us know the result ;)

cpboyd, 2019-05-22:
@angolini Unfortunately, after a clean Yocto build with the `6.2.4p4.0` driver, I'm still seeing a seg fault with Xserver 1.20.1 in thud:
```
[     8.748] (EE) Backtrace:
[     8.748] (EE)
[     8.749] (EE) Segmentation fault at address 0x4
[     8.749] (EE)
Fatal server error:
[     8.749] (EE) Caught signal 11 (Segmentation fault). Server aborting
```

Is there a new version of [xf86-video-imx-vivante](https://source.codeaurora.org/external/imx/xf86-video-imx-vivante/)?  The last commit seems to be the same as is currently in `xf86-video-imx-vivante_6.2.4.p1.8.bb`

angolini, 2019-05-22:
I don't think there is a newer version. I think from this point we need to debug it

cpboyd, 2019-05-23:
@angolini Here's my logs after two separate clean Yocto builds (both with the latest Vivante driver):
[xorg1.19.log](https://github.com/Freescale/meta-freescale/files/3212484/xorg1.19.log)
[xorg1.20.log](https://github.com/Freescale/meta-freescale/files/3212485/xorg1.20.log)

A few notes:
This message appears in the working Xorg 1.19 log, but not 1.20:
```
[     6.963] (II) VIVANTE(0): RandR 1.2 enabled, ignore the following RandR disabled message.
```

In 1.20, there's a `Initializing extension RANDR` message after the driver fails (the `Initializing extension` messages didn't appear in 1.19, perhaps because Vivante's DRI completed successfully).

Also in 1.20, there was this extra message after `Using gamma correction` which seems like it could also foretell of problems:
```
[     6.999] (DB) xf86MergeOutputClassOptions unsupported bus type 0
```

MaxKrummenacher, 2019-06-22:
Variscite has a fix:
https://github.com/varigit/meta-variscite-fslc/commit/d51c571a44feca398cebb218cfe9c187976cb5be
which actually reverts this oe commit:
https://git.openembedded.org/openembedded-core/commit/?id=cde11398e6d74ad8f27334199b4bd99cdf1f0ff7
For me this fixes the segfault.

otavio, 2019-06-22:
@MaxKrummenacher mind to prepare a PR?

MaxKrummenacher, 2019-06-24:
Voilà!

I noticed that NXP now put the same fix into their BSP (4.19.35), so I backported from that.

cpboyd, 2019-06-24:
@MaxKrummenacher, thanks!

I hadn't kept up with Variscite's latest changes.

closed 2019-06-24 by cpboyd