Freescale/meta-freescale #109: do_populate_sdk: No manifest generated from: lib32-alsa-lib

manszigher, 2019-06-06:
Hi,

I am trying to build an SDK I suspect the same error would pop up when building an image. The problem that I am facing is that I need to include lib32-alsa-lib to my SDK but I am getting the following error

> ERROR: strix-sdk-1.0-r0 do_populate_sdk: No manifest generated from: lib32-alsa-lib in virtual:multilib:lib32:/home/builder/workdir/layers/poky/meta/recipes-multimedia/alsa/alsa-lib_1.1.5.bb
ERROR: strix-sdk-1.0-r0 do_populate_sdk: Function failed: do_populate_sdk
ERROR: Logfile of failure stored in: /home/builder/workdir/builds/imx8mm/tmp/work/imx8mm-strix-linux/strix-sdk/1.0-r0/temp/log.do_populate_sdk.138
ERROR: Task (/home/builder/workdir/layers/meta-strix/meta-strix-common/recipes-core/images/strix-sdk.bb:do_populate_sdk) failed with exit code '1'

The reason for this is that that PACKAGE_ARCH is overwritten at

https://github.com/Freescale/meta-freescale/blob/c927ec08f6853311b835defc3fa3179fef5de2c3/recipes-multimedia/alsa/alsa-lib_%25.bbappend#L10

and that the find_sstate_manifest function in poky

https://github.com/kraj/poky/blob/2b8d484f9623b9e998d642cff44c09dcf4a0b2b8/meta/lib/oe/sstatesig.py#L579

will not look for the manifest named armv7ahf-neon-mx8mm set by

https://github.com/Freescale/meta-freescale/blob/c927ec08f6853311b835defc3fa3179fef5de2c3/recipes-multimedia/alsa/alsa-lib_%25.bbappend#L10

Do you have any idea how I can solve this based on the code in

https://github.com/kraj/poky/blob/2b8d484f9623b9e998d642cff44c09dcf4a0b2b8/meta/lib/oe/sstatesig.py#L580

It should be possible to add armv7ahf-neon-mx8mm to PACKAGE_EXTRA_ARCHS but I cannot find how I should do that.

BR
Måns Zigher

otavio, 2019-06-06:
We need to adjust it for the target package. What branch are you using?

manszigher, 2019-06-11:
Hi,

I am using rocko. I solved it by adding the following line to our machine config

> PACKAGE_EXTRA_ARCHS = "${PACKAGE_EXTRA_ARCHS_tune-${DEFAULTTUNE}} armv7ahf-neon-mx8mm" 

But I am not sure about this solution.

manszigher, 2019-06-11:
Could it be that there is an issue with my machine resulting in the MACHINE_SOARCH not being included in the list of pkgarcs in https://github.com/kraj/poky/blob/2b8d484f9623b9e998d642cff44c09dcf4a0b2b8/meta/lib/oe/sstatesig.py#L579?

otavio, 2020-05-06:
Inactive, closing.

closed 2020-05-06 by otavio