Freescale/meta-freescale #111: gstreamer1.0-plugins-base 1.14.imx  undefined reference to `pow@GLIBC_2.29'

NguyenTrongThinh, 2019-06-18:
Branch: THUD
OS: Ubuntu 18.04
Machine: iMX8mq-evk

ERROR:

| /home/thinhnguyen/DATA2/EVK/Yocto/build-EVK/tmp/work/aarch64-mx8m-evk-linux/gstreamer1.0-plugins-base/1.14.imx-r0/recipe-sysroot-native/usr/bin/aarch64-evk-linux/../../libexec/aarch64-evk-linux/gcc/aarch64-evk-linux/8.2.0**/ld:** /home/thinhnguyen/DATA2/EVK/Yocto/build-evk/tmp/work/aarch64-mx8m-evk-linux/gstreamer1.0-plugins-base/1.14.imx-r0/recipe-sysroot/usr/lib/**libgstreamer-1.0.so: undefined reference to `pow@GLIBC_2.29'**
| collect2: error: ld returned 1 exit status
| Makefile:717: recipe for target 'gst-discoverer-1.0' failed
| make[2]: *** [gst-discoverer-1.0] Error 1
| make[2]: Leaving directory '/home/thinhnguyen/DATA2/EVK/Yocto/build-evk/tmp/work/aarch64-mx8m-evk-linux/gstreamer1.0-plugins-base/1.14.imx-r0/build/tools'
| Makefile:718: recipe for target 'all-recursive' failed
| make[1]: *** [all-recursive] Error 1
| make[1]: Leaving directory '/home/thinhnguyen/DATA2/EVK/Yocto/build-evk/tmp/work/aarch64-mx8m-evk-linux/gstreamer1.0-plugins-base/1.14.imx-r0/build'
| Makefile:649: recipe for target 'all' failed
| make: *** [all] Error 2
| ERROR: oe_runmake failed

angolini, 2019-06-27:
let me test the build to try to reproduce

angolini, 2019-06-30:
I have not faced this error. I'm also using 18.04, and I set DISTRO= fslc-xwayland

Could you, please, repo sync and try again?

NguyenTrongThinh, 2019-07-01:
after synchronising, I still got the same error. Which branch that you tested? thud or master?

NguyenTrongThinh, 2019-07-04:
Because I downgrade from Warrior to Thud so gstreamer link to glibc2.29 instead of 2.28. Delete tmp folder them rebuild solve the issues.

closed 2019-07-04 by NguyenTrongThinh