Freescale/meta-freescale #115: mesa 19.1.0 doesn't build

fbertux, 2019-06-19:
Hi @thochstein!

Mesa 19.1.0 doesn't compile without adding _use-mainline-bsp_ override, it fails with error:

```shell
Host machine cpu family: arm
Host machine cpu: arm
Target machine cpu family: arm
Target machine cpu: arm
Build machine cpu family: x86_64
Build machine cpu: x86_64

meson.build:422:4: ERROR: Problem encountered: building dri drivers require at least one windowing system or classic osmesa
```

When using _use-mainline-bsp_ build works fine.

Thanks!

otavio, 2019-06-19:
@thochstein this is a side effect of new mesa releases.

angolini, 2019-06-30:
Hi @fbertux can you, please, share the step-by-step to face the same error you've seen?

Which branch, and which local.conf additional line have you used?

fbertux, 2019-07-01:
Hi @angolini !

I'm using master branch with imx6qdlsabresd machine and fslc-framebuffer distro. I just run ```bitbake mesa``` and error shows up.

angolini, 2019-07-01:
Thanks @fbertux. I will work on this build tomorrow

angolini, 2019-07-02:
I have faced the very same error on my side.

zandrey, 2019-08-23:
Hello all!

@angolini: I've faced the same issue recently. I tried to dig it down and came to the following points below. They might be partially wrong, as I am not anywhere near to be a Mesa expert, but they seems logical to me.

- Since DRI drivers are enabled in the build, there should be a backend defined which should have those provided;
- This support is included via gallium when _use-mainline-bsp_ is set, which also enables the Etnaviv support;

When _use-mainline-bsp_ is not set: with the [Mesa commit], it is required that if DRI is set and swrast is to be used - osmesa should be set to "classic" build. PACKAGECONFIG[osmesa] block is defined on OE master and defaults to `-Dosmesa=none`, which causes the build error for DRI drivers. [This OE master commit] has this configuration introduced, which is to me a proper way of handling dependency but somehow causes the _osmesa_ PACKAGECONFIG option to be undefined and falls back to `-Dosmesa=none` switch used in meson build.

At the current moment I am not sure whether i.MX8M Mini GPU drivers do provide DRI interface and are integrated in gallium.

Therefore, I ended up with switching to swrast and disabling gallium in my build, effectively using the classic OSMesa. With this setup the build runs through.  I know this is not the most performant configuration, therefore I would need someone with Mesa expertise to have a look at the proper configuration.

I've prepared the [patch in my fork] that you can have a look at to see which settings I've used to have this solved. RFC is really welcomed here! And if that commit makes either no sense or making things worse - I'd be glad to hear about that as well.

-- andrey

[Mesa commit]: https://gitlab.freedesktop.org/mesa/mesa/commit/431e9abaaba6386aa7fbc1ec0e2566a3f8999f5d
[This OE master commit]: http://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/meta/recipes-graphics/mesa/mesa.inc?id=e47715708729fe3fed2c5bcafa06f5f1c84f302f
[patch in my fork]: https://github.com/zandrey/meta-freescale/commit/4384ef70f079d32f523aa7412e033d3f98de9f0a

zandrey, 2019-08-23:
@angolini: I saw that @otavio have just merged my mesa-related commit in PR #151 together with another change. This should solve the mesa build issue (at least for imx8m mini machine), but should be addressed further with Mesa experts, as the renderer is switched to swrast (software-based) which might have a performance impact.

@otavio: I'm sorry for that, didn't realize that commit would be added to PR. Should've done that on a separate branch... I leave it up to you whether you would want to keep it in `master` of revert.

-- andrey

otavio, 2019-08-23:
One major issue is that NXP does not support framebuffer anymore so it is a general issue. @zandrey fixing the build is critical and the performance can be fixed next.

fschrempf, 2020-02-13:
Actually Etnaviv+Gallium seems to work for i.MX8MM with a mainline BSP, but the change mentioned above makes it impossible to reenable the Gallium drivers in another bbappend. I therefore changed this to make it possible to switch back to the previous implementation if needed. See https://github.com/Freescale/meta-freescale/pull/287.

otavio, 2020-04-23:
Seems all good now, closing this.

closed 2020-04-23 by otavio