Freescale/meta-freescale #135: Got an error “Checking for program pkg-config : not found” during execution “bitbake image_name”

unclesav, 2019-07-23:
I built the image with **gstreamer-imx** plugin and bitbake asked me for **imx-codec**. I found it in **meta-freescale** layer and added it into my image.

Now my **local.conf** has the next additional lines:
_IMAGE_INSTALL_append = " gstreamer1.0-plugins-imx imx-codec"
ACCEPT_FSL_EULA = "1"_

And **bblayers.conf**:
_BBLAYERS  = "\
  ${OEROOT}/../meta-gstreamer1.0 \
  ${OEROOT}/../meta-nodejs \
  ${OEROOT}/../meta-phytec \
  ${OEROOT}/../meta-qt5 \
  ${OEROOT}/../meta-rauc \
  ${OEROOT}/../meta-yogurt \
  ${OEROOT}/../meta-freescale \
  "_

And during the build I have got the next issue (on screenshots)
https://i.imgur.com/2aEUJKq.png
https://i.imgur.com/VQOqOhJ.png

I have checked the version of installed pkg-config and it's env variable:
```
echo $PKG_CONFIG_PATH
 /usr/local/lib/x86_64-linux-gnu/pkgconfig
```

```
pkg-config --version 
 0.29.1
```

Does anyone know how ti avoid this trouble?
I am working with phyboard-nunki (IMX6)

angolini, 2019-08-13:
I am not familiar with meta-gstreamer1.0  or meta-pythec.

Have you tried to get some support from pythec?

otavio, 2020-04-23:
Not related here, closing the issues.

closed 2020-04-23 by otavio