Freescale/meta-freescale #138: gstreamer-plugins-good_1.14.imx compilation error

QB4-dev, 2019-07-24:
`gstreamer-plugins-good` v4l2 plugin compilation error for iMX6ULL:
```
../../../git/sys/v4l2/gstv4l2videodec.c: In function 'gst_v4l2_video_dec_loop':
| ../../../git/sys/v4l2/gstv4l2videodec.c:645:30: error: 'DRM_FORMAT_MOD_AMPHION_TILED' undeclared (first use in this function); did you mean 'DRM_FORMAT_MOD_VIVANTE_TILED'?
|        guint64 drm_modifier = DRM_FORMAT_MOD_AMPHION_TILED;
|                               ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
|                               DRM_FORMAT_MOD_VIVANTE_TILED

| Makefile:928: recipe for target 'libgstvideo4linux2_la-gstv4l2videodec.lo' failed
| make[3]: *** [libgstvideo4linux2_la-gstv4l2videodec.lo] Error 1

| Makefile:898: recipe for target 'v4l2' failed
| make[2]: *** [v4l2] Error 2

```

angolini, 2019-08-15:
I could build fsl-image-multimedia on master today without error.

Could you, please, elaborate what is the configuration you've been using?

QB4-dev, 2019-12-04:
Sorry for late answer. 
I was using weird custom distro probably forcing to use vivante GPU which is not present in iMX6ULL.
All bulids now without errors with fsl-image-multimedia from meta-freescale-distro

closed 2019-12-04 by QB4-dev