Freescale/meta-freescale #140: openssl-qoriq shared library packaging problems

dkl, 2019-07-29:
Hi,

The openssl-qoriq recipe puts libcrypto.so.1.1/libssl.so.1.1 into both libcrypto/libssl packages and into the openssl-qoriq-ptest package. This is a problem because Yocto detects multiple shlib providers for libcrypto.so.1.1 and libssl.so.1.1 and can end up using the wrong one as dependency for other packages using these shared libraries.

If openssl-qoriq-ptest is used as the dependency for everything, then libcrypto/libssl are not even included in the rootfs image, so no programs using openssl will work because libcrypto.so.1.1/libssl.so.1.1 exist only in /usr/lib/openssl-qoriq/ptest/ instead of /lib or /usr/lib.

This issue is made worse due to a bug in package.bbclass which has indeterministic behaviour in case of multiple shlib providers, see this upstream fix:
https://git.yoctoproject.org/cgit/cgit.cgi/poky/commit/?id=41a5dbd16b0c9f5f97e7a160830cf7ca5de52ec6

Looking at openssl-qoriq_1.1.0g.bb, do_install_ptest() already contains some code which possibly was intended to prevent this issue (by renaming the *.so symlinks), but looks like it's incomplete (not handling the real *.so.* files).

Our work-around for now is to disable ptest completely (remove it from DISTRO_FEATURES) so openssl-qoriq-ptest doesn't get built at all.

otavio, 2020-04-23:
@guochunrong please check this.

otavio, 2020-05-06:
@dkl we did merge a lot of changes on dunfell, please take a look and reopen if need.

closed 2020-05-06 by otavio

dkl, 2020-05-08:
OK, I will keep an eye on this, especially when upgrading to dunfell.