Freescale/meta-freescale #179: libimxdmabuffer build error

tyama-zaki, 2019-10-02:
I appended libimxdmabuffer in yocto image.

The following error occurred : 

ERROR: libimxdmabuffer-1.0.0+gitAUTOINC+db17cb57d1-r0 do_install_ptest_base: Function failed: do_install_ptest_base (log file is located at /data/imx-yocto-bsp/build-xwayland/tmp/work/aarch64-mx8mm-poky-linux/libimxdmabuffer/1.0.0+gitAUTOINC+db17cb57d1-r0/temp/log.do_install_ptest_base.13630)
ERROR: Logfile of failure stored in: /data/imx-yocto-bsp/build-xwayland/tmp/work/aarch64-mx8mm-poky-linux/libimxdmabuffer/1.0.0+gitAUTOINC+db17cb57d1-r0/temp/log.do_install_ptest_base.13630
Log data follows:
| DEBUG: Executing shell function do_install_ptest_base
| grep: Makefile: No such file or directory
| install: cannot stat '/data/imx-yocto-bsp/build-xwayland/tmp/work/aarch64-mx8mm-poky-linux/libimxdmabuffer/1.0.0+gitAUTOINC+db17cb57d1-r0/git/test-alloc': No such file or directory
| WARNING: exit code 1 from a shell command.
| ERROR: Function failed: do_install_ptest_base (log file is located at /data/imx-yocto-bsp/build-xwayland/tmp/work/aarch64-mx8mm-poky-linux/libimxdmabuffer/1.0.0+gitAUTOINC+db17cb57d1-r0/temp/log.do_install_ptest_base.13630)
ERROR: Task (/data/imx-yocto-bsp/sources/meta-freescale/recipes-bsp/libimxdmabuffer/libimxdmabuffer_1.0.0.bb:do_install_ptest_base) failed with exit code '1'
NOTE: Tasks Summary: Attempted 4332 tasks of which 4331 didn't need to be rerun and 1 failed.

When I checked the work folder, it seemed that the path was specified incorrectly.

libimxdmabuffer_1.0.0.bb
    install -m 0755 ${B}/test-alloc ${D}${PTEST_PATH}/test
I changed the above line to:
    install -m 0755 ${B}/test/test-alloc.c ${D}${PTEST_PATH}/test
libimxdmabuffer build success.

I would like to hear if this change is okay.

MaxKrummenacher, 2019-10-06:
Hi

${B}/test/test-alloc is a binary providing unit tests to OE's ptest system.
It is compiled from ${S}/test/test-alloc.c, compare with https://github.com/Freescale/libimxdmabuffer/blob/master/wscript#L304

From your error output:
    | install: cannot stat '.../git/test-alloc': No such file or directory
The file should actually be taken from '.../build/test-alloc', so somehow your setup does have the variable B strangely set.

Did you do a cleanup and rebuild, e.g.

    MACHINE=imx8mmevk bitbake libimxdmabuffer -c cleansstate; MACHINE=imx8mmevk bitbake libimxdmabuffer

Note that for me the above runs through without any error on a recent checkout of master.

Max

otavio, 2020-04-22:
@dv1 please take a look.

dv1, 2020-05-06:
Please retry. I completely reworked the way ptest is implemented.

otavio, 2020-05-06:
I think we can close this as it is inactive for a while. If happens again we can reopen.

closed 2020-05-06 by otavio