Freescale/meta-freescale #184: DP/eDP port support on imx8mq

agherzan, 2019-10-14:
I saw that the current kernel revision for imx8mq seems to have support for DP in `imx-hdp` driver. There is also a dp/edp device tree. Have you guys ever tested anything in that regard? I was looking for a board with DP but couldn't find any.

otavio, 2020-04-22:
We did not. @thochstein is NXP aware of any?

thochstein, 2020-04-29:
We're working on a possibility. @agherzan, in the meantime please let us know if your status changes.

agherzan, 2020-04-29:
Since our initial request, we switched to HDMI for the time being but we are interested if that would be a possible in the future.

otavio, 2021-04-10:
@thochstein news on this? Otherwise we could close this issue.

thochstein, 2021-04-13:
No news. I asked again a couple of days ago but no response. I guess you can close it.

closed 2021-04-13 by otavio