Freescale/meta-freescale #187: qtbase: update patches

cpboyd, 2019-10-23:
Originally filed as #42, but closed without comment.

0016-Configure-eglfs-with-egl-pkg-config.patch hasn't been needed since Qt v5.9:
https://github.com/qt/qtbase/commit/e9a7739e77f79baeb5452189b6d17e63de5b341d

In Yocto Warrior and Zeus, this causes a fuzz warning:
```
WARNING: qtbase-5.13.0+gitAUTOINC+fc9ae22c88-r0 do_patch: Fuzz detected:

Applying patch 0016-Configure-eglfs-with-egl-pkg-config.patch
patching file src/plugins/platforms/eglfs/eglfs-plugin.pro
Hunk #1 succeeded at 7 with fuzz 2 (offset 5 lines).

The context lines in the patches can be updated with devtool:

    devtool modify qtbase
    devtool finish --force-patch-refresh qtbase <layer_path>

Don't forget to review changes done by devtool!

WARNING: qtbase-5.13.0+gitAUTOINC+fc9ae22c88-r0 do_patch: QA Issue: Patch log indicates that patches do not apply cleanly. [patch-fuzz]
```

otavio, 2019-10-23:
Mind to prepare a PR fixing this?

cpboyd, 2019-10-23:
@otavio Sure.  Do I need to submit two separate ones for `master` and `warrior` branches?

otavio, 2019-10-23:
> @otavio Sure. Do I need to submit two separate ones for `master` and `warrior` branches?

Prepare for master; then we make a new one for warrior once it is merged.

PR #188 (2019-10-23): fix(qtbase): remove unnecessary patch
Fixes #187 

The old patch was rendered unnecessary by this commit https://github.com/qt/qtbase/commit/e9a7739e77f79baeb5452189b6d17e63de5b341d on qtbase.
closed 2019-10-23 by otavio