Freescale/meta-freescale #192: kernel-module-imx-gpu-viv compile error with Yocto Zeus and linux-fslc-imx 4.9-1.0.x

dv1, 2019-11-03:
The same problem from https://github.com/Freescale/meta-freescale/issues/136 also appears when trying to build kernel 4.9 with Yocto Zeus:

I used this workaround:

    diff --git a/hal/os/linux/kernel/gc_hal_kernel_driver.c b/hal/os/linux/kernel/gc_hal_kernel_driver.c
    index 87f26e721773..52f829bc477e 100644
    --- a/hal/os/linux/kernel/gc_hal_kernel_driver.c
    +++ b/hal/os/linux/kernel/gc_hal_kernel_driver.c
    @@ -1284,7 +1284,7 @@ static struct platform_driver gpu_driver = {
    };
    #endif /* USE_LINUX_PCIE */
    
    -static int __init gpu_init(void)
    +static int gpu_init(void)
    {
        int ret = 0;
    
    @@ -1315,7 +1315,7 @@ static int __init gpu_init(void)
        return 0;
    }
    
    -static void __exit gpu_exit(void)
    +static void gpu_exit(void)
    {
    #if USE_LINUX_PCIE
        pci_unregister_driver(&gpu_driver);

But backporting the proper patches would fix this issue in Zeus builds.

otavio, 2019-11-04:
@thochstein mind to take a look?

thochstein, 2019-11-04:
Will do.

thochstein, 2019-11-14:
Sorry, I got pulled away from this. What kernel are you building? I'm not able to duplicate this building with linux-imx_4.14.98.

thochstein, 2019-11-15:
Note @otavio , the issue was reported against zeus. Will you handle that? Or do you want me to cherry pick?

closed 2019-11-18 by otavio

zandrey, 2019-11-19:
Hello guys!

Those patches are actually came into v4.4.188: [fe5844365ec6c4d61838f289926f4d55da94d2fb ](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=fe5844365ec6c4d61838f289926f4d55da94d2fb) and [2c34c215c1028f971af7cf8189c4f6237afdb90b](https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/patch/?id=2c34c215c1028f971af7cf8189c4f6237afdb90b)

@thochstein I can make a PR with those patches instead - would be easier to track when linux-fslc-imx_4.9-1.0.x gets bumped up to 4.4.188. If you don't mind then I can do it.

-- andrey

zandrey, 2019-11-19:
> Hello guys!
>
> Those patches are actually came into v4.4.188: [fe5844365ec6c4d61838f289926f4d55da94d2fb [...]

For sure I meant 4.9.188! Links pointing to correct patches.

thochstein, 2019-11-19:
Sounds good to me. Thanks @zandrey!

PR #201 (2019-11-18): linux-fslc-imx_4.9-1.0.x: Backport patches to fix building with gcc 9
Closes #192.
commit de48050f (2019-11-18): linux-fslc-imx_4.9-1.0.x: Backport patches to fix building with gcc 9

Closes #192.