Freescale/meta-freescale #229: Revert "linux-qoriq recipes: reorder merge_delta_config task"

rehsack, 2019-12-13:
6e0b6f unfortunately causes a blank build for e.g. ls2088ardb fail as that:

```
ERROR: linux-qoriq-4.19-r0 do_preconfigure: Execution of '/build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614' failed with exit code 2:
/build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: 119: /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: cannot open /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/defconfig: No such file
WARNING: exit code 2 from a shell command.

ERROR: Logfile of failure stored in: /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/log.do_preconfigure.13614
Log data follows:
| DEBUG: Executing shell function do_preconfigure
| /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: 119: /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: cannot open /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/defconfig: No such file
| WARNING: exit code 2 from a shell command.
| ERROR: Execution of '/build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614' failed with exit code 2:
| /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: 119: /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/temp/run.do_preconfigure.13614: cannot open /build/sno/mdr-gt-build/tmp/work/ls2088ardb-poky-linux/linux-qoriq/4.19-r0/defconfig: No such file
| WARNING: exit code 2 from a shell command.
|
ERROR: Task (/build/sno/sources/meta-freescale/recipes-kernel/linux/linux-qoriq_4.19.bb:do_preconfigure) failed with exit code '1'
```

This is caused by do_preconfigure() from classes/fsl-kernel-localversion.bbclass expects a ${WORKDIR}/defconfig which was earlier created by merge_delta_config.

I have no idea what linux-variscite is which @zpfvo has problems with, but I think we should find a better solution.

It seems, that preconfigure has to be run after merge_delta_config and we should figure out why this linux-variscite fails.

otavio, 2019-12-13:
Please send a PR for it, so we merge it.

rehsack, 2019-12-13:
I was unsure whether immediately revert is best reaction. I wanted @zpfvo having a chance to jump in ;)
I can send PR in 2-3 days anyway ;)

zpfvo, 2019-12-15:
Hi,

please revert if there are problems. The recipe from variscite I was
referring to is here:
https://github.com/varigit/meta-variscite-fslc/blob/thud/recipes-kernel/linux/linux-variscite_4.14.78.bb
.

Ill try to figure it out why this is working for me. Its totally possible
that the error is caused by the adaption of variscite and does not lie
within this layer.

closed 2019-12-16 by otavio