Freescale/meta-freescale #242: Weston does not start after upgrade (imx8mmevk machine with 4.19.y kernel)

zandrey, 2019-12-19:
Duplicating the original thread from _meta-freescale_ yocto mailing list:

After the recent upgrade of Weston (commit: a4fdb4074644bda09b46490163886065b4c00aab) I started to receive the error from G2D on i.MX8M Mini at the start with following output:
```
Date: 2019-12-18 UTC
[16:51:24.982] weston 6.0.1
               https://wayland.freedesktop.org
               Bug reports to:
https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 6.0.1-36-g4ee0a1da-dirty g2d-renderer: refine
enable overlay view feature (2019-10-17 12:57:52 +0800)
[16:51:24.982] Command line: /usr/bin/weston
[16:51:24.982] OS: Linux, 4.19.35-1.1.0+g0f9917c56d59, #1 SMP PREEMPT
Wed Oct 30 11:02:38 UTC 2019, aarch64
[16:51:24.982] Using config file '/etc/xdg/weston/weston.ini'
[16:51:24.982] Output repaint window is 16 ms maximum.
[16:51:24.982] Loading module '/usr/lib/libweston-6/drm-backend.so'
[16:51:24.986] initializing drm backend
[16:51:24.991] logind: session control granted
[16:51:24.994] using /dev/dri/card0
[16:51:24.994] DRM: supports universal planes
[16:51:24.994] DRM: supports atomic modesetting
[16:51:24.994] DRM: supports picture aspect ratio
[16:51:24.994] Loading module '/usr/lib/libweston-6/g2d-renderer.so'
[16:51:24.995] g2d_open fail.
[16:51:24.995] g2d_renderer_create faile.
[16:51:24.995] failed to initialize g2d render
```

If the G2D support is disabled in the _/etc/xdg/weston/weston.ini_ by commenting the line:
`#use-g2d=1`

then Weston falls back to use the GL backend, which also fails with the following output:
```
Date: 2019-12-18 UTC
[16:56:14.652] weston 6.0.1
               https://wayland.freedesktop.org
               Bug reports to:
https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 6.0.1-36-g4ee0a1da-dirty g2d-renderer: refine
enable overlay view feature (2019-10-17 12:57:52 +0800)
[16:56:14.653] Command line: /usr/bin/weston
[16:56:14.653] OS: Linux, 4.19.35-1.1.0+g0f9917c56d59, #1 SMP PREEMPT
Wed Oct 30 11:02:38 UTC 2019, aarch64
[16:56:14.653] Using config file '/etc/xdg/weston/weston.ini'
[16:56:14.653] Output repaint window is 16 ms maximum.
[16:56:14.653] Loading module '/usr/lib/libweston-6/drm-backend.so'
[16:56:14.657] initializing drm backend
[16:56:14.663] logind: session control granted
[16:56:14.667] using /dev/dri/card0
[16:56:14.667] DRM: supports universal planes
[16:56:14.667] DRM: supports atomic modesetting
[16:56:14.667] DRM: supports picture aspect ratio
[16:56:14.672] Loading module '/usr/lib/libweston-6/gl-renderer.so'
[16:56:14.704] EGL client extensions: EGL_EXT_client_extensions
               EGL_EXT_platform_base EGL_KHR_platform_wayland
               EGL_EXT_platform_wayland EGL_KHR_platform_gbm
[16:56:14.712] EGL version: 1.5
[16:56:14.712] EGL vendor: Vivante Corporation
[16:56:14.712] EGL client APIs: OpenGL_ES OpenVG
[16:56:14.712] EGL extensions: EGL_KHR_fence_sync EGL_KHR_reusable_sync
               EGL_KHR_wait_sync EGL_KHR_image EGL_KHR_image_base
               EGL_KHR_image_pixmap EGL_KHR_gl_texture_2D_image
               EGL_KHR_gl_texture_cubemap_image EGL_KHR_gl_renderbuffer_image
               EGL_EXT_image_dma_buf_import
               EGL_EXT_image_dma_buf_import_modifiers EGL_KHR_lock_surface
               EGL_KHR_create_context EGL_KHR_surfaceless_context
               EGL_EXT_create_context_robustness EGL_EXT_buffer_age
               EGL_ANDROID_native_fence_sync EGL_WL_bind_wayland_display
               EGL_WL_create_wayland_buffer_from_image EGL_KHR_partial_update
[16:56:14.712] EGL_KHR_surfaceless_context available
[16:56:14.725] failed to create context
[16:56:14.725] EGL error state: EGL_BAD_ALLOC (0x3003)
[16:56:14.725] EGL error state: EGL_SUCCESS (0x3000)
[16:56:14.725] failed to initialize egl
```

`zeus` branch is not affected by this issue, there the Weston is operable.

Following are SRCREVs of components in question that could've potentially impact the Weston.
From `master` branch build (which is failing):
```
# imx8mmevk-fslc-linux
SRCREV_pn-optee-os = "6a52487eb0ff664e4ebbd48497f0d3322844d51d"
SRCREV_pn-kernel-module-qca6174 = "2f4ae376ce3db65c3270a4bf7e804f3e86052c8c"
SRCREV_pn-imx-boot = "dd0234001713623c79be92b60fa88bc07b07f24f"
SRCREV_pn-linux-imx = "0f9917c56d5995e1dc3bde5658e2d7bc865464de"
SRCREV_pn-u-boot-imx = "4d377539a1190e838eae5d8b8a794dde0696d572"
SRCREV_pn-kernel-module-qca9377 = "2f4ae376ce3db65c3270a4bf7e804f3e86052c8c"
SRCREV_pn-kernel-module-imx-gpu-viv = "0f9917c56d5995e1dc3bde5658e2d7bc865464de"
SRCREV_pn-cryptodev-module = "fd8b15ef1c8398a69a37932ee48c74ab40329a29"

# aarch64-mx8mm-fslc-linux
SRCREV_pn-gstreamer1.0-plugins-good = "a31aabb7581d3f491cd31889d44479bb0f34990b"
SRCREV_pn-imx-gpu-apitrace = "09579e67262af9c993dd082055a924c2c61cf34d"
SRCREV_pn-imx-gst1.0-plugin = "0c5f48b3aee473d8c8f3e853eb2078f054caa4d2"
SRCREV_pn-gstreamer1.0-plugins-bad = "0191521ba226904e4b2f84c38e5f6ae75169a18a"
SRCREV_pn-imx-alsa-plugins = "cde60d68ab2acee913dbfacb8aabb53d87dd3e38"
SRCREV_boostdesc_pn-opencv = "34e4206aef44d50e6bbcd0ab06354b52e7466d26"
SRCREV_contrib_pn-opencv = "2c32791a9c500343568a21ea34bf2daeac2adae7"
SRCREV_ipp_pn-opencv = "32e315a5b106a7b89dbed51c28f8120a48b368b4"
SRCREV_opencv_pn-opencv = "371bba8f54560b374fbcd47e7e02f015ac4969ad"
SRCREV_vgg_pn-opencv = "fccf7cd6a4b12079f73bbfb21745f9babcd4eb1d"
SRCREV_pn-gstreamer1.0 = "d42548da09724ad8cc1aa4f1944607920be2f4c0"
SRCREV_pn-linux-imx-headers = "a51dda8feec03e8c675230786d64ee050944dbfd"
SRCREV_pn-imx-gpu-sdk = "d35bac9419895ea516c25e2f36a6084729d9e6ce"
SRCREV_pn-imx-atf = "70fa7bcc1a2035ab8402550911b3ae29eff55371"
SRCREV_pn-weston = "4ee0a1da010053b35495e82ad30aa5a8b1c34e6c"
SRCREV_pn-libdrm = "5748c8ff40f1ae87487c01e580f145a43542cbda"
SRCREV_pn-gstreamer1.0-plugins-base = "f62f8ab685e549eefaecdb1e700239a5a408fcad"
SRCREV_pn-imx-test = "709d14b037e7a45af018875c4a880f8f9c198288"
```

From `zeus` branch build (which is operable):
```
# imx8mmevk-fslc-linux
SRCREV_pn-optee-os = "6a52487eb0ff664e4ebbd48497f0d3322844d51d"
SRCREV_pn-kernel-module-qca6174 = "e008d2578b09f1f593f65ede48529a52312e3957"
SRCREV_pn-imx-boot = "dd0234001713623c79be92b60fa88bc07b07f24f"
SRCREV_pn-linux-imx = "1175b59611537b0b451e0d1071b1666873a8ec32"
SRCREV_pn-u-boot-imx = "0d6d8807793e74402e23772062469aeb4808dd48"
SRCREV_pn-kernel-module-qca9377 = "e008d2578b09f1f593f65ede48529a52312e3957"
SRCREV_pn-kernel-module-imx-gpu-viv = "f54dc1486d13d44766cdb6551d094313f077b535"
SRCREV_pn-cryptodev-module = "fd8b15ef1c8398a69a37932ee48c74ab40329a29"

# aarch64-mx8mm-fslc-linux
SRCREV_pn-gstreamer1.0-plugins-good = "a31aabb7581d3f491cd31889d44479bb0f34990b"
SRCREV_pn-imx-gpu-apitrace = "013499c63d9eddd8c792bd8b136991caa3b8a100"
SRCREV_pn-imx-gst1.0-plugin = "0c5f48b3aee473d8c8f3e853eb2078f054caa4d2"
SRCREV_pn-gstreamer1.0-plugins-bad = "0191521ba226904e4b2f84c38e5f6ae75169a18a"
SRCREV_pn-imx-alsa-plugins = "9a63071e7734bd164017f3761b8d1944c017611f"
SRCREV_boostdesc_pn-opencv = "34e4206aef44d50e6bbcd0ab06354b52e7466d26"
SRCREV_contrib_pn-opencv = "2c32791a9c500343568a21ea34bf2daeac2adae7"
SRCREV_ipp_pn-opencv = "32e315a5b106a7b89dbed51c28f8120a48b368b4"
SRCREV_opencv_pn-opencv = "371bba8f54560b374fbcd47e7e02f015ac4969ad"
SRCREV_vgg_pn-opencv = "fccf7cd6a4b12079f73bbfb21745f9babcd4eb1d"
SRCREV_pn-gstreamer1.0 = "d42548da09724ad8cc1aa4f1944607920be2f4c0"
SRCREV_pn-linux-imx-headers = "dfc78102d29f81d9316b281a3b8af3dff95174c8"
SRCREV_pn-imx-gpu-sdk = "d35bac9419895ea516c25e2f36a6084729d9e6ce"
SRCREV_pn-imx-atf = "1cb68fa0a0dd8bc00b9871b51d4c4e1d0a827b2d"
SRCREV_pn-weston = "fb563901657b296c7c7c86d26602a622429e334f"
SRCREV_pn-libdrm = "95645843f59495387a072d48374718f22e69d7a4"
SRCREV_pn-gstreamer1.0-plugins-base = "f62f8ab685e549eefaecdb1e700239a5a408fcad"
```
To me it looks like graphics drivers are broken in the 4.19.y version (which were also recently upgraded to match kernel major/minor).

Looks like we're hitting a regression in Weston with all recent upgrades from NXP of all components (Kernel, drivers, Weston, ...).

@otavio @thochstein @mihaix: can you please have a look at this?
I believe a lot of people would be affected by this once Yocto 3.1 would roll-out and everyone would
start to upgrade.

Thanks a lot!

-- andrey

zandrey, 2019-12-19:
Some additional Weston logging information from the image built from `zeus` branch:
```
Date: 2019-12-19 UTC
[08:14:43.775] weston 5.0.0
               https://wayland.freedesktop.org
               Bug reports to: https://gitlab.freedesktop.org/wayland/weston/issues/
               Build: 5.0.0-33-gfb563901-dirty MGS-4627 [#ccc] weston will turn off when panel-psotion=none and use gplay (2019-03-14 11:30:30 +0800)
[08:14:43.775] Command line: /usr/bin/weston --log=/var/log/weston.log
[08:14:43.775] OS: Linux, 4.14.98-imx+g1175b5961153, #1 SMP PREEMPT Thu Dec 19 08:36:24 UTC 2019, aarch64
[08:14:43.776] Using config file '/etc/xdg/weston/weston.ini'
[08:14:43.779] Output repaint window is 16 ms maximum.
[08:14:43.780] Loading module '/usr/lib/libweston-5/drm-backend.so'
[08:14:43.787] initializing drm backend
[08:14:43.787] logind: not running in a systemd session
[08:14:43.787] logind: cannot setup systemd-logind helper (-61), using legacy fallback
[08:14:43.791] using /dev/dri/card0
[08:14:43.791] DRM: supports universal planes
[08:14:43.791] DRM: supports atomic modesetting
[08:14:43.791] DRM: does not support picture aspect ratio
[08:14:43.802] Loading module '/usr/lib/libweston-5/g2d-renderer.so'
[08:14:43.922] event0  - 30370000.snvs:snvs-powerkey: is tagged by udev as: Keyboard
[08:14:43.922] event0  - 30370000.snvs:snvs-powerkey: device is a keyboard
[08:14:43.925] event3  - Chicony HP Business Slim Keyboard: is tagged by udev as: Keyboard
[08:14:43.925] event3  - Chicony HP Business Slim Keyboard: device is a keyboard
[08:14:43.929] event4  - Chicony HP Business Slim Keyboard: is tagged by udev as: Keyboard
[08:14:43.929] event4  - Chicony HP Business Slim Keyboard: device is a keyboard
[08:14:43.931] event1  - gpio_ir_recv: is tagged by udev as: Keyboard
[08:14:43.931] event1  - gpio_ir_recv: device is a keyboard
[08:14:43.933] event2  - MCE IR Keyboard/Mouse (gpio-rc-recv): is tagged by udev as: Keyboard Mouse
[08:14:43.933] event2  - MCE IR Keyboard/Mouse (gpio-rc-recv): device is a pointer
[08:14:43.933] event2  - MCE IR Keyboard/Mouse (gpio-rc-recv): device is a keyboard
[08:14:43.987] DRM: head 'HDMI-A-1' found, connector 33 is connected, EDID make 'SAM', model 'SMS24A650', serial 'HTPC800767'
[08:14:43.987] Registered plugin API 'weston_drm_output_api_v1' of size 24
[08:14:43.996] No backlight control for output 'HDMI-A-1'
[08:14:43.996] Output HDMI-A-1 (crtc 31) video modes:
               1920x1080@60.0, preferred, current, 148.5 MHz
               1920x1080@59.9, 148.4 MHz
               1920x1080@50.0, 148.5 MHz
               1600x1200@60.0, 162.0 MHz
               1680x1050@59.9, 119.0 MHz
               1280x1024@75.0, 135.0 MHz
               1280x1024@60.0, 108.0 MHz
               1440x900@75.0, 136.8 MHz
               1440x900@59.9, 88.8 MHz
               1280x960@60.0, 108.0 MHz
               1280x800@59.9, 71.0 MHz
               1152x864@75.0, 108.0 MHz
               1280x720@60.0, 74.2 MHz
               1280x720@59.9, 74.2 MHz
               1280x720@50.0, 74.2 MHz
               1024x768@75.0, 78.8 MHz
               1024x768@70.1, 75.0 MHz
               1024x768@60.0, 65.0 MHz
               832x624@74.6, 57.3 MHz
               800x600@75.0, 49.5 MHz
               800x600@72.2, 50.0 MHz
               800x600@60.3, 40.0 MHz
               800x600@56.2, 36.0 MHz
               720x576@50.0, 27.0 MHz
               720x480@60.0, 27.0 MHz
               720x480@59.9, 27.0 MHz
               640x480@75.0, 31.5 MHz
               640x480@72.8, 31.5 MHz
               640x480@66.7, 30.2 MHz
               640x480@60.0, 25.2 MHz
               640x480@59.9, 25.2 MHz
               720x400@70.1, 28.3 MHz
[08:14:43.997] associating input device event0 with output HDMI-A-1 (none by udev)
[08:14:43.997] associating input device event3 with output HDMI-A-1 (none by udev)
[08:14:43.997] associating input device event4 with output HDMI-A-1 (none by udev)
[08:14:43.997] associating input device event1 with output HDMI-A-1 (none by udev)
[08:14:43.997] associating input device event2 with output HDMI-A-1 (none by udev)
[08:14:43.997] Output 'HDMI-A-1' enabled with head(s) HDMI-A-1
[08:14:43.997] Compositor capabilities:
               arbitrary surface rotation: no
               screen capture uses y-flip: no
               presentation clock: CLOCK_MONOTONIC, id 1
               presentation clock resolution: 0.000000001 s
[08:14:44.003] Loading module '/usr/lib/weston/desktop-shell.so'
[08:14:44.006] launching '/usr/libexec/weston-keyboard'
[08:14:44.008] launching '/usr/libexec/weston-desktop-shell'
[08:14:44.362] deactivating session
[08:14:44.362] event0  - 30370000.snvs:snvs-powerkey: device removed
[08:14:44.410] event3  - Chicony HP Business Slim Keyboard: device removed
[08:14:44.454] event4  - Chicony HP Business Slim Keyboard: device removed
[08:14:44.474] event1  - gpio_ir_recv: device removed
[08:14:44.530] event2  - MCE IR Keyboard/Mouse (gpio-rc-recv): device removed
```

Here is could be seen that Weston is starting perfectly fine.

zandrey, 2019-12-19:
Some more information:

I've ran G2D test application provided by `imx-gpu-g2d` and it is also appears to be broken. The result of execution of `/opt/g2d_samples/g2d_test` is:
```
g2d_open: fail with status -7
g2d_open fail.
```

It seems to me that commit f4d05aa18c0192fea2eef3fbfb3390bdb5033c64 brought a faulty version of G2D, which in turn breaks the Weston by failing to initialize.

Since the `imx-gpu-g2d` is a binary package - there is not much room for further investigation left to be performed, and I would have to relay it further to NXP for investigation...

-- andrey

mihaix, 2019-12-19:
@zandrey thanks for the details. I'm working on upgrading the recipe for imx-gpu-viv, meanwhile, can you give it a try? See my fork.

zandrey, 2019-12-20:
@mihaix thanks for the update!

I've tried your fork and it looks like Weston and G2D are fully operable!

I've seen that the difference is only 2 commits [fae1d46f and 9ce3a55e] from the `master` of your fork; the rest looks aligned with `master` branch of `meta-freescale`.

There is a QA hiccup in your fork that should be addressed:
```
ERROR: imx-gpu-viv-1_6.4.0.p1.0-aarch64-r0 do_package_qa: QA Issue: libgbm-imx rdepends on imx-gpu-viv-dev [dev-deps]
ERROR: imx-gpu-viv-1_6.4.0.p1.0-aarch64-r0 do_package_qa: QA Issue: libegl-imx rdepends on imx-gpu-viv-dev [dev-deps]
ERROR: imx-gpu-viv-1_6.4.0.p1.0-aarch64-r0 do_package_qa: QA run found fatal errors. Please consider fixing them.
```

For my personal build to run through, I've patched it with following statements:
```
+INSANE_SKIP_libegl-imx += "dev-so dev-deps"
+INSANE_SKIP_libgbm-imx += "dev-deps"
```

This is however not a preferred way, as it is quite unclear why `ligegl` and `libgbm` require a -dev package from `imx-gpu-viv`... It might be that since do_package_qa is executed in parallel and this recipe provides both libegl and libgbm - the QA kicks a bit earlier than expected, which causes this issue...

Nevertheless, I would be waiting now till you'd make a PR and test the layer itself after it would be merged.

Thanks a lot!

-- andrey

mihaix, 2019-12-20:
Thank you for the update, that's great!
You're right, that's not a pretty way of fixing those QA issues, I've seen those later and was still working on them. I've updated the commit, it should be fine now. There were newly added libraries that were not covered in any of the packages and ended up in imx-gpu-viv-dev, which led to the dev-deps QA issue.

Thanks,
-- Mihai

zandrey, 2019-12-20:
> Thank you for the update, that's great!

Thanks for solving this issue pretty quick! :)

> You're right, that's not a pretty way of fixing those QA issues, I've seen those later and was still working on them. I've updated the commit, it should be fine now. There were newly added libraries [...]

True, this totally slipped my mind: this is actually a default behavior. If nothing consumes files - they are landing in `-dev` package and leading to QA Error afterwards. I'd note this for myself for the future.

Anyways, I guess once the PR would come - this issue can be closed.

-- andrey

zandrey, 2019-12-23:
Hello all,

I've tested the latest `master` and with PR #249 this issue is resolved, Weston is fully operable.

Thanks for taking care of this.

-- andrey

closed 2019-12-23 by zandrey