Freescale/meta-freescale #253: imx8m Nano: missing machine description

zandrey, 2020-01-06:
Hello all,

I've started to work on the build for i.MX8M Nano CPU and found out that actually the `master` branch has a machine definition missing.

There are binaries available from NXP website, which could be obtained under the [Linux Binary Demo Files - i.MX 8M Nano EVK](https://www.nxp.com/webapp/Download?colCode=L4.14.98_2.2.0_MX8MN&appType=license) and it looks like the image is based on the `sumo` branch. I've checked this repo, but unfortunately was not able to find any reference to the _imx8mnevk_ machine.

Does this machine description exists somewhere and can someone from NXP contribute it here?

I've created a draft machine description for my testing purposes but faced an issue with the U-Boot not starting: SPL uses the ROMAPI to boot the image, but loading from SD interface fails. I was actually hoping that if the machine description would be contributed - I could sort out the issue with U-Boot I'm currently having.

Thanks a lot!

-- andrey

otavio, 2020-01-06:
You could possibly send your draft and then we mark @thochstein on the issue, so he can comment on it and we can them upstream it.

zandrey, 2020-01-06:
@otavio That would be great! I'd prepare a PR then in my fork to link it here.

otavio, 2020-01-06:
Cool. Thanks for the help.

mihaix, 2020-01-06:
> @otavio That would be great! I'd prepare a PR then in my fork to link it here.

CC: @panfir 

@zandrey machine conf for _imx8mnevk_ is available in _meta-fsl-bsp-release_ [1] layer, it needs to be pulled in and adapted here.

[1] https://source.codeaurora.org/external/imx/meta-fsl-bsp-release/tree/imx/meta-bsp/conf/machine/imx8mnevk.conf?h=warrior-4.19.35-1.1.0

zandrey, 2020-01-06:
@mihaix Thanks for your clarification here! I've not seen your comments and already came up with the PR #254 here.

@otavio Can you please hold off the #254? I'd have a look at what is available at the link @mihaix provided above and would amend the PR accordingly.

Thank you all for such a prompt support!

-- andrey

zandrey, 2020-01-07:
Hello all,
PR #254 is updated and boot issue is solved now. Once it is merged - this issue could be closed.

-- andrey

closed 2020-01-09 by zandrey