Freescale/meta-freescale #256: xstart: Segmentation fault at address 0x4

cyrilfr, 2020-01-07:
I still experiment #93 with x11-base IMAGE_FEATURE on a UDOO Neo.
I try to run the [Electron](https://electronjs.org/) framework on the official st1232 7'' touchscreen kit for UDOO Neo.

cyrilfr, 2020-02-17:
UP!

cyrilfr, 2020-03-06:
According to [this thread](https://community.nxp.com/thread/526149), the imx-gpu-viv 6.2.4 driver must be avoided. Is there a way to use imx-gpu-viv 5.0.11 instead on thud?

cyrilfr, 2020-04-15:
I applied [this patch](https://github.com/Freescale/meta-freescale/commit/b504c047698c4b74a859c5f23d654f348449c179) and now it works.

otavio, 2020-04-15:
@cyrilfr please open a PR applying b504c047698c4b74a859c5f23d654f348449c179 to thud, so we can close this issue.

cyrilfr, 2020-04-15:
I'll do it as soon as possible :)
However, it does fix the segmentation fault and display something but my st1232 7'' touchscreen touch events doesn't work.

otavio, 2020-04-15:
This is another issue. So fix one at the time.

closed 2020-04-15 by cyrilfr