Freescale/meta-freescale #280: u-boot-fslc fails for imx5 machines

twoerner, 2020-02-11:
building on master:

        Build Configuration:
        BB_VERSION           = "1.44.0"
        BUILD_SYS            = "x86_64-linux"
        NATIVELSBSTRING      = "opensuseleap-15.1"
        TARGET_SYS           = "arm-oe-linux-gnueabi"
        MACHINE              = "imx53qsb"
        DISTRO               = "nodistro"
        DISTRO_VERSION       = "nodistro.0"
        TUNE_FEATURES        = "arm armv7a vfp thumb neon callconvention-hard"
        TARGET_FPU           = "hard"
        meta-freescale       = "master:9227f4c9e6d77d3fa15f67d178ba9da02c5550fe"
        meta                 = "master:44a4ac2294da0f53cbbfabc7ece836fe97f4d3f7"

I get:

        | make[1]: *** No rule to make target '/z/build-master/meta-freescale/build/tmp-glibc/work/imx53qsb-oe-linux-gnueabi/u-boot-fslc/v2019.07+gitAUTOINC+ca0ab15271-r0/git/arch//cpu/u-boot.lds', needed by 'u-boot.lds'.  Stop.

and also:

        |       echo >&2 "  /z/build-master/meta-freescale/build/tmp-glibc/work/imx53qsb-oe-linux-gnueabi/u-boot-fslc/v2019.07+gitAUTOINC+ca0ab15271-r0/git is not clean, please run 'make mrproper'"; \

$ bitbake virtual/bootloader -e | grep "^B="
 24 B="/z/build-master/meta-freescale/build/tmp-glibc/work/imx53qsb-oe-linux-gnueabi/u-boot-fslc/v2019.07+gitAUTOINC+ca0ab15271-r0/git"
 25 $ bitbake virtual/bootloader -e | grep "^S="
 26 S="/z/build-master/meta-freescale/build/tmp-glibc/work/imx53qsb-oe-linux-gnueabi/u-boot-fslc/v2019.07+gitAUTOINC+ca0ab15271-r0/git"

twoerner, 2020-02-11:
ps I'm also seeing this with imx233-olinuxino-maxi (meta-freescale-3rdparty)

texierp, 2020-02-11:
Hi @twoerner,

Seems related to this commit in oe-core http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-bsp/u-boot?h=master-next&id=26023b6b0f897842fd98b3e10a8acd5b3ad8f418

otavio, 2020-02-11:
> Hi @twoerner,
> 
> Seems related to this commit in oe-core http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-bsp/u-boot?h=master-next&id=26023b6b0f897842fd98b3e10a8acd5b3ad8f418

You seem right @texierp. @twoerner can you take a look how we can make a generic fix?

twoerner, 2020-02-11:
> @twoerner can you take a look how we can make a generic fix?

okay. i'm looking at it now

otavio, 2020-04-22:
This was already fixed. Closing...

closed 2020-04-22 by otavio