Freescale/meta-freescale #292: libdrm is broken

emantec1, 2020-02-29:
The codeaurora.org git appears to no longer be available resulting in building libdrm to fail

ERROR: libdrm-2.4.91.imx-r0 do_fetch: Fetcher failure for URL: 'git://source.codeaurora.org/external/imx/libdrm-imx.git;protocol=https;branch=libdrm-imx-2.4.91'. Unable to fetch URL from any source.
ERROR: libdrm-2.4.91.imx-r0 do_fetch: Function failed: base_do_fetch

mihaix, 2020-03-02:
It's there. Try checking your network connection / proxy settings.
Or try again maybe it was some temporary downtime on codeaurora.

otavio, 2020-04-22:
It seems no one complained anymore, closing it.

otavio, 2021-10-08:
@thochstein

otavio, 2021-10-11:
@iamshamiullah please get the `log.do_fetch` so it provides more information how it is failing.

angolini, 2022-02-01:
Close due to inactivity.

closed 2022-02-01 by angolini