Freescale/meta-freescale #316: build fails on master after commit split out u-boot tools

nicola-lunghi, 2020-04-01:
after commit dad59d2dd0bb42ca5a8c3c1c697c3576b76d334c and 24f414f37128b5c23b93bbe35a2eb20163988325 the build fails

bitbake core-image-minimal
ERROR: ParseError at /home/user/build/proj/newgw/meta-freescale/recipes-bsp/u-boot/u-boot-imx-tools_2019.04.bb:1: Could not include required file recipes-bsp/u-boot/u-boot-tools.inc

otavio, 2020-04-01:
You're likely using master branch with other OE-Core branches. What are your branches?

nicola-lunghi, 2020-04-01:
yes sorry you are probably right I'll double check with the latest master

zandrey, 2020-04-03:
@otavio, I have the same issue with building `u-boot-imx` from master of _oe-core_ and _meta-freescale_

Following are the errors I get from build:
```
| make: Entering directory '/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/git'
| make -C /work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/build/imx8mm_evk_config KBUILD_SRC=/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/git \
| -f /work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/git/Makefile u-boot-initial-env
| make[1]: Entering directory '/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/build/imx8mm_evk_config'
| make[1]: *** No rule to make target 'u-boot-initial-env'.  Stop.
| make[1]: Leaving directory '/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/build/imx8mm_evk_config'
| Makefile:148: recipe for target 'sub-make' failed
| make: *** [sub-make] Error 2
| make: Leaving directory '/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/git'
| WARNING: exit code 1 from a shell command.
| 
ERROR: Task (/work/development/yocto-master/yocto-build-appliance/meta-freescale/recipes-bsp/u-boot/u-boot-imx_2019.04.bb:do_compile) failed with exit code '1'
```

This has been caused by your commit [4882a93566f2ef24787777d33950f12a9cc5db4e](https://git.yoctoproject.org/cgit.cgi/poky/commit/?id=4882a93566f2ef24787777d33950f12a9cc5db4e) on `master`. If reverted - the build runs through.

Can you suggest what additional changes should be done to `u-boot-imx_2019.04.bb` recipe so the _u-boot-initial-env_ target gets built?

Thanks!

-- andrey

otavio, 2020-04-03:
> @otavio, I have the same issue with building `u-boot-imx` from master of _oe-core_ and _meta-freescale_

This is not the same issue.

> Following are the errors I get from build:
>
> ```
> | make: Entering directory '/work/development/yocto-master/build-output/work/imx8mmevk-fsl-linux/u-boot-imx/2019.04-r0/git'
> | make -C [...]

This seems to be due to `u-boot-imx` fork to not provide this target. 

Checking the source this target was introduced on 2019.07. 

Please apply this and see if it works: https://github.com/u-boot/u-boot/commit/bdaa73a5b392.patch

If it does, prepare a PR with it please.

zandrey, 2020-04-03:
@otavio Great catch! 😉 I've applied the patch, and the build now succeeds!

PR is on the way.

-- andrey

nicola-lunghi, 2020-04-03:
Yes thanks build with latest poky

closed 2020-04-03 by nicola-lunghi