Freescale/meta-freescale #320: Error: OpenCl support on imx8mq processor with vivante GPU

Christian-Prather, 2020-04-02:
Hello, we are currently working on a project that requires GPU parallel processing on an **imx8mq** processor with a **vivante** GPU. We have tried building for the imx8 evaluation kit as well as for our own carrier board and were unable to successfully get OpenCl to populate onto the build. We were able to get OpenCv however when attempting to detect OpenCl it was not on the image. We also compiled the SDK for this particular build which contained the needed OpenCl libraries for our architecture though when attempting to run any OpenCL program after transferring them to our image they were not able to execute. We are wondering if there is a way to get OpenCl working using just the meta-freescale layer. Thank you in advance for any assistance you can offer!

Machine: imx8mq

Branch: zeus

otavio, 2020-04-03:
@thochstein please take a look.

thochstein, 2020-04-03:
Hi @Christian-Prather, probably the image you create does not contain any apps that depend on OpenCL. You can add it explicitly from `local.conf`:

`IMAGE_INSTALL_append = " libopencl-imx"`

That won't be necessary if you want to configure OpenCV with OpenCL support. I was able to build opencv after adding this to `local.conf`:

```
PACKAGECONFIG_append_pn-opencv = " opencl"
PREFERRED_PROVIDER_opencl-headers_imxgpu     ?= "imx-gpu-viv"
```

Christian-Prather, 2020-04-09:
Thank you this seemed to do the trick!

closed 2020-04-10 by Christian-Prather

thochstein, 2020-04-10:
Hey, glad it worked!