Freescale/meta-freescale #331: iMX8MQ based board stopped booting the Linux kernel on zeus

mirzak, 2020-04-12:
I noticed that my iMX8MQ based board stopped booting the Linux kernel on zeus branch after #326 was merged.

The error I see during boot is the following:

```
[    4.469525] i.mx8-hdp 32c00000.hdmi: 0,ff,ff,ff,ff,ff,ff,0                                 
[   26.045415] INFO: rcu_preempt detected stalls on CPUs/tasks:                               
[   26.056537]  2-...: (2 GPs behind) idle=cfc/0/0 softirq=515/515 fqs=0                      
[   26.068567]  3-...: (0 ticks this GP) idle=610/0/0 softirq=481/481 fqs=0                   
[   26.080898]  (detected by 1, t=5422 jiffies, g=-160, c=-161, q=71)                         
[   26.092642] Task dump for CPU 2:                                                           
[   26.101373] swapper/2       R  running task        0     0      1 0x00000002               
[   26.114005] Call trace:                                                                    
[   26.121953] [<ffff000008085b54>] __switch_to+0x94/0xd8                                     
[   26.132648] [<ffff000008094fe0>] arm_cpuidle_suspend+0x30/0x38                             
[   26.144061] [<ffff0000089f59b0>] arm_enter_idle_state+0x38/0x68                            
[   26.155539] [<ffff0000089f3a38>] cpuidle_enter_state+0x1c8/0x228                           
[   26.167090] [<ffff0000089f3ad0>] cpuidle_enter+0x18/0x20                                   
[   26.177864] [<ffff000008108ea8>] call_cpuidle+0x18/0x30                                    
[   26.188486] [<ffff0000081090e0>] do_idle+0x1a0/0x1d8                                       
[   26.198805] [<ffff0000081092c4>] cpu_startup_entry+0x24/0x28                               
[   26.209856] [<ffff00000808f098>] secondary_start_kernel+0x110/0x120                        
[   26.221516] Task dump for CPU 3:                                                           
[   26.230093] swapper/3       R  running task        0     0      1 0x00000002               
[   26.242574] Call trace:                                                                    
[   26.250329] [<ffff000008085b54>] __switch_to+0x94/0xd8                                     
[   26.260880] [<ffff000008094fe0>] arm_cpuidle_suspend+0x30/0x38                             
[   26.272115] [<ffff0000089f59b0>] arm_enter_idle_state+0x38/0x68                            
[   26.283477] [<ffff0000089f3a38>] cpuidle_enter_state+0x1c8/0x228                           
[   26.294966] [<ffff0000089f3ad0>] cpuidle_enter+0x18/0x20                                   
[   26.305703] [<ffff000008108ea8>] call_cpuidle+0x18/0x30                                    
[   26.316314] [<ffff0000081090e0>] do_idle+0x1a0/0x1d8                                       
[   26.326519] [<ffff0000081092c4>] cpu_startup_entry+0x24/0x28                               
[   26.337308] [<ffff00000808f098>] secondary_start_kernel+0x110/0x120                        
[   26.348638] rcu_preempt kthread starved for 5422 jiffies! g18446744073709551456 c1844674407
3709551455 f0x0 RCU_GP_WAIT_FQS(3) ->state=0x402 ->cpu=3                                      
[   26.367215] rcu_preempt     I    0     8      2 0x00000000                                 
[   26.378046] Call trace:                                                                    
[   26.385763] [<ffff000008085b54>] __switch_to+0x94/0xd8                                     
[   26.396294] [<ffff000008d46620>] __schedule+0x1a8/0x590                                    
[   26.406893] [<ffff000008d46a40>] schedule+0x38/0xa0                                        
[   26.417126] [<ffff000008d49d10>] schedule_timeout+0x178/0x318                              
[   26.428277] [<ffff000008127c58>] rcu_gp_kthread+0x3f8/0x7b0                                
[   26.439234] [<ffff0000080e9c10>] kthread+0x100/0x130                                       
[   26.449545] [<ffff000008084df0>] ret_from_fork+0x10/0x18       
```

mirzak, 2020-04-12:
Isolated to the following commit: https://github.com/Freescale/meta-freescale/commit/54d14f17ebd961dc7cb9ef89b76613218fb7562d

mirzak, 2020-04-12:
My build configuration,

```
Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-fslc-linux"
MACHINE              = "coral-dev"
DISTRO               = "fslc-xwayland"
DISTRO_VERSION       = "3.0"
TUNE_FEATURES        = "aarch64"
TARGET_FPU           = ""
meta                 
meta-poky            = "HEAD:5531ffc5668c2f24b9018a7b7174b5c77315a1cf"
meta-oe              
meta-multimedia      
meta-python          = "HEAD:9e60d30669a2ad0598e9abf0cd15ee06b523986b"
meta-freescale       = "zeus:5fdac18a1f4daed214e07af014d068b7b6196f5c"
meta-freescale-3rdparty = "HEAD:bebf0ad8a4a95f657f93e96a2bd0b1046617e990"
meta-freescale-distro = "HEAD:ca27d12e4964d1336e662bcc60184bbff526c857"
meta-clang           = "zeus:81ba160c95b12b2922f99b60bef25ab37a5e2f0e"
meta-coral-bsp       = "zeus:448c000cd3e4514a412bf789a209ffbf908c5110"
```

otavio, 2020-04-12:
@chrisdimich any idea?

chrisdimich, 2020-04-12:
We tested on Nitrogen8M which is iMX8MQ based and we do not see this issue.

otavio, 2020-04-13:
@chrisdimich please help on this as this is a regression of a previously working device. Maybe there is a required change on U-Boot or Linux?

chrisdimich, 2020-04-13:
Sure, we will look at this first thing tomorrow. Again, our validation was obviously using our own Nitrogen boards, so we did not see this issue.

mirzak, 2020-04-13:
I can confirm that I was able to reproduce this on the [imx8mqevk](https://github.com/Freescale/meta-freescale/blob/master/conf/machine/imx8mqevk.conf) target as well.

otavio, 2020-04-13:
@chrisdimich @gibsson please take a look. I wonder how your board is working, have you guys added any change?

gibsson, 2020-04-13:
My suspicion is that ATF has changed his way of doing IRQs in 4.19.
Most likely @chrisdimich didn't update U-Boot from the image. If you keep the U-Boot from before, it will work.
Since U-Boot is in the fastboot partition in our case, you can flash the image without affecting U-Boot nor need to update it.
@mirzak , first can you confirm that you updated U-Boot after flashing the image on Nitrogen8M?
If so, as a test and to confirm my theory, can you flash your nitrogen8m with the following U-Boot: http://linode.boundarydevices.com/u-boot-images/u-boot.nitrogen8m
If this recipe was updated only for Nano support, either revert to previous version (and we'll remove bootloader from nano conf for now) or use v4.14.98_2.3.0 instead (the one we use right now with 4.14 kernel).

mirzak, 2020-04-13:
> @mirzak , first can you confirm that you updated U-Boot after flashing the image on Nitrogen8M?

I do not have a Nitrogen8M board, I have tested on imx8mqevk. I can confirm that I flash U-Boot as I am I am booting from SD card and using the wic image to flash it, e.g

```
zcat /tmp/core-image-base-imx8mqevk.wic.gz | sudo dd of=/dev/sda bs=8M
```

gibsson, 2020-04-13:
Oh sorry I thought you saw that issue on Nitrogen8M at first, no problem.
So yeah it seems you can't mix 4.19 atf with older kernels, nice.
I'll let @chrisdimich decide what he wants to do for nano but I'd suggest reverting both ATF and OPTEE updates in zeus. Sorry for the inconvenience.

otavio, 2020-04-13:
@chrisdimich please send the reverts.

chrisdimich, 2020-04-13:
Sure will do and as Gary mentioned we will just remove bootloader from nano conf for now.

chrisdimich, 2020-04-13:
Reverted and sent PR #332

otavio, 2020-04-13:
Fixed in #332.

closed 2020-04-13 by otavio