Freescale/meta-freescale #334: cairo doesn't compile

cyrilfr, 2020-04-14:
cairo doesn't compile on "master". The do_compile tasks gives an error:

| arm-poky-linux-gnueabi-libtool: link: arm-poky-linux-gnueabi-gcc -march=armv7-a -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a9 -fstack-protector-strong -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot -D_REENTRANT -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/pixman-1 -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/freetype2 -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/freetype2 -DLINUX -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/libdrm -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/libdrm -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/libpng16 -Wall -Wextra -Wmissing-declarations -Werror-implicit-function-declaration -Wpointer-arith -Wwrite-strings -Wsign-compare -Wpacked -Wswitch-enum -Wmissing-format-attribute -Wvolatile-register-var -Wstrict-aliasing=2 -Winit-self -Wunsafe-loop-optimizations -Wno-missing-field-initializers -Wno-unused-parameter -Wno-attributes -Wno-long-long -Winline -fno-strict-aliasing -fno-common -Wp,-D_FORTIFY_SOURCE=2 -Wno-unused-but-set-variable -D_REENTRANT -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/include/glib-2.0 -I/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/lib/glib-2.0/include -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0=/usr/src/debug/cairo/1.14.12-r0 -fdebug-prefix-map=/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot= -fdebug-prefix-map=/poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot-native= -Wl,-O1 -Wl,--hash-style=gnu -Wl,--as-needed -fstack-protector-strong -Wl,-z -Wl,relro -Wl,-z -Wl,now -o .libs/cairo-sphinx cairo_sphinx-sphinx.o  ../../util/cairo-script/.libs/libcairo-script-interpreter.so ../../boilerplate/.libs/libcairoboilerplate.a /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/build/src/.libs/libcairo.so ../../src/.libs/libcairo.so -lpthread -lpixman-1 -lfontconfig -lfreetype -lGLESv2 -lEGL -ldl -lpng16 -lxcb-shm -lxcb -lxcb-render -lXrender -lX11 -lXext -lz -lglib-2.0 -lrt -lm
| /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/8.2.0/ld: /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/lib/libGAL.so: undefined reference to `log@GLIBC_2.29'
| /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/8.2.0/ld: /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/lib/libGAL.so: undefined reference to `pow@GLIBC_2.29'
| /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot-native/usr/bin/arm-poky-linux-gnueabi/../../libexec/arm-poky-linux-gnueabi/gcc/arm-poky-linux-gnueabi/8.2.0/ld: /poky/build-neo-thud/tmp/work/cortexa9t2hf-neon-mx6sx-poky-linux-gnueabi/cairo/1.14.12-r0/recipe-sysroot/usr/lib/libGAL.so: undefined reference to `exp@GLIBC_2.29'
| collect2: error: ld returned 1 exit status

zandrey, 2020-04-14:
`cairo` on oe-core *master* is at version `1.16.0`:
```
meta/recipes-graphics/cairo/cairo_1.16.0.bb
```

From your log, looks like you're trying to build old version (`1.14.12`), which is not `master` branch.

Can you share your build information?

-- andrey

cyrilfr, 2020-04-14:
I'm currently on `thud` in poky. What I've seen is that the pre-built libGAL.so file in imx-gpu-viv has been compiled with glibc 2.29 (and newer??) as dependency (using `readelf`). Poky in `thud` uses glibc 2.28.

I try to build using master because X11 crashes in thud. Maybe my strategy is wrong. I've seen that [this commit](https://github.com/Freescale/meta-freescale/commit/b504c047698c4b74a859c5f23d654f348449c179) fixes a crash of X11 but I'm not sure if it's related.

zandrey, 2020-04-14:
In this case you should try to build the entire image of `master` branch (including *poky*). I've just tried it for `imx8mmevk` machine and *cairo* builds just fine.

-- andrey

cyrilfr, 2020-04-14:
I plan to upgrade systems already running images from `thud`. This is not possible.

cyrilfr, 2020-04-15:
Please close this issue.

closed 2020-04-15 by otavio

zandrey, 2020-04-15:
@cyrilfr Have you found a solution? If you did - can you please share a brief here, so others who might have the same trouble can refer here.

> 
> Please close this issue.

You can do it yourself, but you've already figured it out. 😉

cyrilfr, 2020-04-15:
I just moved back to `thud` and added a `xserver-xorg_%.bbappend` file to my machine layer.
```

FILESEXTRAPATHS_prepend := "${THISDIR}/${BPN}:"
SRC_URI_append_imxgpu2d = " file://0003-Remove-check-for-useSIGIO-option.patch"
```
and added [this patch](https://github.com/Freescale/meta-freescale/commit/b504c047698c4b74a859c5f23d654f348449c179) to the recipe files in `xserver-xorg` directory (like in meta-freescale `master`).

Having both poky and meta-freescale on `thud` makes glibc version fits the requirements for cairo and imx-gpu-viv.

otavio, 2020-04-15:
@zandrey he did find the commit to backport. He mentioned it on #256. I asked him to prepare a PR with the backport fix.