Freescale/meta-freescale #349: Integrate i.MX release L5.4.3-1.0.0

thochstein, 2020-04-17:


thochstein, 2020-04-17:
I'm working on recipes-graphics.

otavio, 2020-04-17:
@thochstein good; please send PRs as soon as something is read.
@zandrey what are you working on now?
@panfir and you?

zandrey, 2020-04-17:
> 
> @zandrey what are you working on now?

I was playing with the `linux-fslc` repo and wanted to try out the `5.4.3-1.0.0-imx` branch on the `imx8mmevk` machine. If it turns out to be OK - I would come up with the recipe to pull this kernel into the layer (in addition to official from NXP).

> 
> 
> I'm working on recipes-graphics.

@thochstein are you planning to upgrade the ` meta-freescale-distro/recipes-graphics/imx-gpu-sdk/imx-gpu-sdk_5.0.2.bb` as well? I saw that there is an updated version in your BSP: `meta-sdk/recipes-graphics/imx-gpu-sdk/imx-gpu-sdk_5.4.0.bb`. If not - I might take a look at it.

Another thing I've noted during the upgrade: commit IDs for Kernel recipe, Linux Headers and Vivante kernel module do not match in NXP BSP:
```
meta-bsp/recipes-kernel/linux/linux-imx_5.4.bb -> f8118585ee3c7025265b28985fdfe0af96a84466
meta-bsp/recipes-kernel/linux/linux-imx-headers_5.4.bb -> e5569d9e14060947f28f0783e41615f1007f1094
meta-bsp/recipes-kernel/kernel-modules/kernel-module-imx-gpu-viv_6.4.0.p2.0.bb -> f2af3a66108b88aedb77016e8b59bab425816200
```
@thochstein @panfir is it intended to be like this? If so - should we adjust the versions in this layer? If I understood it correct, the only commit which is tagged with `lf-5.4.y-1.0.0` - is the one from `linux-imx_5.4.bb` recipe, so all other recipes which has the same SRC_URI should use the same commit ID.

-- andrey

otavio, 2020-04-17:
> > @zandrey what are you working on now?
> 
> I was playing with the `linux-fslc` repo and wanted to try out the `5.4.3-1.0.0-imx` branch on the `imx8mmevk` machine. If it turns out to be OK - I would come up with the recipe to pull this kernel into the layer (in addition to official from NXP).

Please do; looking forward to it.

> Another thing I've noted during the upgrade: commit IDs for Kernel recipe, Linux Headers and Vivante kernel module do not match in NXP BSP:
>
> ```
> meta-bsp/recipes-kernel/linux/linux-imx_5.4.bb -> [...]

Kind of. The `linux-imx-headers` do not need to be bumped every time Linux does. Doing so would cause all the recipes to rebuild and if the API/ABI did not change it is not need.

For now, we could use the branch tip so we pick fixes NXP did after their GA release.

thochstein, 2020-04-17:
> @thochstein are you planning to upgrade the ` meta-freescale-distro/recipes-graphics/imx-gpu-sdk/imx-gpu-sdk_5.0.2.bb` as well? I saw that there is an updated version in your BSP: `meta-sdk/recipes-graphics/imx-gpu-sdk/imx-gpu-sdk_5.4.0.bb`. If not - I might take a look at it.

Good point, I updated my comment above to be more specific. If you want to take it, just update the ticket that you will take it? If you don't take it I will do the same if/when I am ready to work on it?

zandrey, 2020-04-17:
> 
> Kind of. The `linux-imx-headers` do not need to be bumped every time Linux does. Doing so would cause all the recipes to rebuild and if the API/ABI did not change it is not need.

Agree. The only thing would be to track changes in ABI/API in this case, but at this point it should be the same. It might deviate later in the future, but with NXP kernel I assume it would stay stable since there are no updates planned.

My major concern was actually about the `kernel-module-imx-gpu-viv_6.4.0.p2.0.bb`. It is built from the same kernel repo, but uses a different commit. I had a look the diff between this commit and tag - they do have differences. It means that we have the kernel built from one commit and module - from another one, which might lead to kernel panics and hiccups.

@thochstein if you don't mind - can you comment on this point? Thanks a lot!

zandrey, 2020-04-17:
> 
> 
> Good point, I updated my comment above to be more specific. If you want to take it, just update the ticket that you will take it? If you don't take it I will do the same if/when I am ready to work on it?

I guess I would deal with it after the validation of `linux-fslc` repo. I would leave this point open for the moment, as I don't know when I would come to it, but chances are I'll do it early next week. 😉 

-- andrey

thochstein, 2020-04-17:
> Agree. The only thing would be to track changes in ABI/API in this case, but at this point it should be the same. It might deviate later in the future, but with NXP kernel I assume it would stay [...]

I didn't like the extra maintenance of maintaining a separate kernel module repo, and so updated our recipe to grab the correct sub-tree of the linux source itself. We update `kernel-module-imx-gpu-viv_6.4.0.p2.0.bb` iff the relevant sub-tree changes, hence the different SHA.

As far as I know it has worked for us, but maybe we have been getting lucky, as I wasn't aware that the commit SHA might cause some problems.

zandrey, 2020-04-17:
> I didn't like the extra maintenance of maintaining a separate kernel module repo, and so updated our recipe to grab the correct sub-tree of the linux source itself. We update [...]

I've just double-checked the source tree diff between kernel and module commits, and in this case it should be fine, no random luck involved. 😄 

If you ask me personally, I would have the following logical chain:

Normally, SHA would not cause any damage - it is more a question of consistency that kernel itself and module are produced from the same source tree using the same SHA. Later during upgrade it would be easier to say, grep `meta-freescale` repo and replace the SHA with the updated one in all recipes.

But if SHA would differ however - then "grepping technique" would not work. So the person doing upgrade should be aware of the fact that module is built from the same source tree and should be updated as well. But even in this case, that developer should know exactly to which SHA the module should be updated to. In case if kernel and module have the same SHA - it's a "no brainer", just use the same for both recipes. But in case if they are already different - developer would struggle to find a proper one.

-- andrey

thochstein, 2020-04-18:
> I've just double-checked the source tree diff between kernel and module commits, and in this case it should be fine, no random luck involved. 😄

Right, or you can do the same thing we do which is to just check the SHA of the base of the interface.

> But if SHA would differ however - then "grepping technique" would not work. So the person doing upgrade should be aware of the fact that module is built from the same source tree and should be [...]

So, we limit SHA updates on our side just to limit rebuilds to only interface changes and not all kernel changes. At the point of upstreaming this is not so important, and we can choose to just sync them. A note in the recipe would make this clear.

thochstein, 2020-04-18:
We upgraded `imx-gpu-apitrace` to 8.0, but the (huge) python 3 patchset that @zandrey did doesn't work on it. We have 9.0 in our trunk, which is the version that adds python 3, and so were thinking to upstream that instead. Prabhu says it should be fine. It's okay?

zandrey, 2020-04-18:
> We upgraded `imx-gpu-apitrace` to 8.0, but the (huge) python 3 patchset that @zandrey did doesn't work on it. 

As far as I remember, I took those patches from the `python3` branch of the official [aiptrace](https://github.com/apitrace/apitrace) repo. According to [tag compare](https://github.com/apitrace/apitrace/compare/8.0...9.0) those patches were applied immediately on top of release 8.0, so they should be OK... 😕 What is the issue with the patch set? Does it not applies? Or the *apitrace* is broken?

> We have 9.0 in our trunk, which is the version that adds python 3, and so were thinking to upstream that instead. Prabhu says it should be fine. It's okay?

I guess it would be fine by me, newer version is usually better! 😃 Besides, all Python 3 patches should be in.

otavio, 2020-04-18:
Yes version 9.0 is fine @thochstein please go for it.

CarolZhuYuqing, 2020-04-20:
Hi Otavio,
I am working on imx vpu/codec/parser/vpuwrap.

panfir, 2020-04-22:
I'm working on jailhouse, xen, m4

zeeblex, 2020-04-22:
Working on imx-kobs

zeeblex, 2020-04-23:
Working on imx-mkimage. Will switch branch to imx_5.4.3_2.0.0 latest commit.

zandrey, 2020-04-23:
> Working on imx-mkimage. Will switch branch to imx_5.4.3_2.0.0 latest commit.

Please use `lf-5.4.y` branch instead, release *2.0.0* is not yet ready and out of scope of this upgrade.

zeeblex, 2020-05-05:
@otavio this means we need to rework the imx-kobs and imx-mkimage to 1.0.0 release(as initially discussed)

zandrey, 2020-05-05:
@thochstein @otavio I've opened a PR #386 with the introduction of new consolidated kernel recipe, and also inheritance on `kernel-yocto` class so we can use config fragment, *scc* files and kernel sanity checks.

I queued it up against `master-next` as the change is pretty intrusive. 😄 

Can you please take a look at it and give your feedback?

Thanks a lot!

-- andrey

thochstein, 2020-05-29:
Working on `qtbase`.

zandrey, 2021-02-08:
@thochstein This issue can be closed I guess, since `dunfell` is on the `2.1.0` NXP release, and `master` already contains `2.3.0`.

Can you close it?

-- andrey

otavio, 2021-02-08:
Yes, this can be closed.

closed 2021-02-08 by otavio