Freescale/meta-freescale #378: dunfell: qtbase configuration issues with EGLFS and FB

cpboyd, 2020-04-30:
I think this might be related to https://github.com/Freescale/meta-freescale/commit/341ba18eccc7703eb5ac23dc900b3bf087d3be20

Basically, I've noticed that in my configuration, it adds `-opengl desktop` rather than `-opengl egl` which throws a configuration error.

cpboyd, 2020-04-30:
After further inspection, the real issue lies with changes to `MACHINEOVERRIDES` https://github.com/Freescale/meta-freescale/blob/dunfell/conf/machine/include/imx-base.inc#L23

Possibly because of the old line related to the `use-mainline-bsp`: https://github.com/Freescale/meta-freescale/blob/dunfell/conf/machine/include/imx-base.inc#L109

cpboyd, 2020-04-30:
Probably related to #86 

May make more sense to modify `MACHINEOVERRIDES_EXTENDER_FILTER_OUT` than to revert the base `MACHINEOVERRIDES` addition of `use-mainline-bsp`, but I'm not familiar with how the `machine-overrides-extender` works

otavio, 2020-04-30:
It does not seem to be the case. I did:

```
% MACHINE=nitrogen6x ye x qtbase PACKAGECONFIG
Loading cache: 100% |#########################################################################################################| Time: 0:00:00
Loaded 3344 entries from dependency cache.
=== Final value
PACKAGECONFIG =      release     accessibility dbus udev evdev widgets tools libs freetype tests pcre                openssl     gles2                         jpeg libpng zlib            eglfs

=== Expansion
PACKAGECONFIG ==>      ${PACKAGECONFIG_RELEASE}     ${PACKAGECONFIG_DEFAULT}     ${PACKAGECONFIG_OPENSSL}     ${PACKAGECONFIG_GL}     ${PACKAGECONFIG_FB}     ${PACKAGECONFIG_X11}     ${PACKAGECONFIG_KDE}     ${PACKAGECONFIG_FONTS}     ${PACKAGECONFIG_SYSTEM}     ${PACKAGECONFIG_DISTRO}  ${PACKAGECONFIG_PLATFORM}
    PACKAGECONFIG_RELEASE ==> release
    PACKAGECONFIG_DEFAULT ==> accessibility dbus udev evdev widgets tools libs freetype tests pcre     ${@bb.utils.contains('SELECTED_OPTIMIZATION', '-Os', 'optimize-size ltcg', '', d)}     ${@bb.utils.contains('DISTRO_FEATURES', 'qt5-static', 'static', '', d)}
        @bb.utils.contains('SELECTED_OPTIMIZATION', '-Os', 'optimize-size ltcg', '', d) ==>
        @bb.utils.contains('DISTRO_FEATURES', 'qt5-static', 'static', '', d) ==>
    PACKAGECONFIG_OPENSSL ==> openssl
    PACKAGECONFIG_GL ==> gles2
    PACKAGECONFIG_FB ==> ${@bb.utils.contains('DISTRO_FEATURES', 'directfb', 'directfb', '', d)}
        @bb.utils.contains('DISTRO_FEATURES', 'directfb', 'directfb', '', d) ==>
    PACKAGECONFIG_X11 ==> ${@bb.utils.contains('DISTRO_FEATURES', 'x11', 'xcb glib xkbcommon', '', d)}
        @bb.utils.contains('DISTRO_FEATURES', 'x11', 'xcb glib xkbcommon', '', d) ==>
    PACKAGECONFIG_KDE ==> ${@bb.utils.contains('DISTRO_FEATURES', 'kde', 'sm cups fontconfig kms gbm libinput sql-sqlite openssl', '', d)}
        @bb.utils.contains('DISTRO_FEATURES', 'kde', 'sm cups fontconfig kms gbm libinput sql-sqlite openssl', '', d) ==>
    PACKAGECONFIG_FONTS ==>
    PACKAGECONFIG_SYSTEM ==> jpeg libpng zlib
    PACKAGECONFIG_DISTRO ==>
    PACKAGECONFIG_PLATFORM ==>      ${@bb.utils.contains('DISTRO_FEATURES', 'x11',     '',        bb.utils.contains('DISTRO_FEATURES', 'wayland', '',                                                        'eglfs', d), d)}
        @bb.utils.contains('DISTRO_FEATURES', 'x11',     '',        bb.utils.contains('DISTRO_FEATURES', 'wayland', '',                                                        'eglfs', d), d) ==> eglfs
```

and more specifically:

```
% MACHINE=nitrogen6x ye x qtbase PACKAGECONFIG_GL
Loading cache: 100% |#########################################################################################################| Time: 0:00:00
Loaded 3344 entries from dependency cache.
=== Final value
PACKAGECONFIG_GL = gles2

=== Expansion
PACKAGECONFIG_GL ==> gles2
```

and as expected, below is for mainline:

```
% MACHINE=imx6qdlsabresd ye x qtbase PACKAGECONFIG_GL
Parsing recipes: 100% |#######################################################################################################| Time: 0:00:13
Parsing of 2245 .bb files complete (0 cached, 2245 parsed). 3344 targets, 485 skipped, 0 masked, 0 errors.
Removing 6 recipes from the x86_64 sysroot: 100% |############################################################################| Time: 0:00:00
=== Final value
PACKAGECONFIG_GL = gl gbm kms

=== Expansion
PACKAGECONFIG_GL ==> ${@bb.utils.contains('DISTRO_FEATURES', 'opengl', 'gl', 'no-opengl', d)} gbm kms
    @bb.utils.contains('DISTRO_FEATURES', 'opengl', 'gl', 'no-opengl', d) ==> gl
```

What layers do you have included and what distro are you using?

cpboyd, 2020-05-01:
@otavio I've been using a BSP based upon the imx6qdlsabreauto/imx6qdlsabresd.

Prior to dunfell, qtbase would build properly.

If you try to build qtbase with framebuffer support (i.e. without X11 or Wayland), the `do_configure` step will fail as PACKAGECONFIG_GL returns gl which corresponds to `-opengl desktop`.

This also seems to prevent using the `imx-gpu-viv` with the "mainline" BSPs. Is that intended?

otavio, 2020-05-01:
Yes, as we default to mainline there. If desired, you can set the BSP to NXP and return to old behavior.

Qt failed to build?

cpboyd, 2020-05-01:
@otavio Yes, Qt fails to build. As noted in #86 previously, the `use-mainline-bsp` filters out `mx6`, `mx6q`, `mx6dl`, etc.

Prior to this change to MACHINEOVERRIDES, `use-mainline-bsp` was only set for specific chipsets, so I guess the `MACHINEOVERRIDES_EXTENDER_FILTER_OUT` wasn't such an issue.

schnitzeltony, 2020-05-02:
Will give dunfell a build and see what happens - wanted to lift our builds anyway. May take a bit.

PR #379 (2020-05-05): Revert "qtbase: Do not pin gles2 when using open source graphic stack"
This reverts commit 1cf9efd20fd94afdd2f7484b0570c570a21b8b65.

The qtbase configure fail to find any suitable GL library, so it fails
with:

,----
| WARNING: No QPA platform plugin enabled! This will produce a Qt that
| cannot run GUI applications.  See "Platform backends" in the output of
| --help.
|
| ERROR: Feature 'opengl-desktop' was enabled, but the pre-condition
| '(config.win32 && !config.winrt && !features.opengles2
|     && (config.msvc || libs.opengl))
|  || (!config.watchos && !config.win32 && !config.wasm && li [... cut]
commit e1b7bce0 (2020-05-05): Revert "qtbase: Do not pin gles2 when using open source graphic stack"

This reverts commit 1cf9efd20fd94afdd2f7484b0570c570a21b8b65.

The qtbase configure fail to find any suitable GL library, so it fails
with:

,----
| WARNING: No QPA platform plugin enabled! This will produce a Qt that
| cannot run GUI applications.  See "Platform backends" in the output of
| --help.
|
| ERROR: Feature 'opengl-desktop' was enabled, but the pre-condition
| '(config.win32 && !config.winrt && !features.opengles2
|     && (config.msvc || libs.opengl))
|  || (!config.watchos && !config.win32 && !config.wasm && l [... cut]
closed 2020-05-05 by otavio