Freescale/meta-freescale #389: Weird behavior using menuconfig with configuration fragments

kevswims, 2020-05-06:
I am trying to use the new support for configuration fragments that was introduced in https://github.com/Freescale/meta-freescale/pull/386. I am following the instructions from the kernel-dev manual here: https://www.yoctoproject.org/docs/latest/kernel-dev/kernel-dev.html#creating-config-fragments to create configuration fragments. I have run the `kernel_configme` step, run the `menuconfig` step and then generated a fragment using `diffconfig`. 

After adding that fragment to my BSP I wanted to make additional changes so I reran the `kernel_configme` step and checked that the `.config` file generated contained the correct changes from my first fragment and it did. However, when I go ahead and run the `menuconfig` step all those changes go away and are not present in the `.config` anymore or in the menuconfig gui. If I load the previous .config into menuconfig then I can see all the old changes I am expecting but then running `diffconfig` creates a fragment with all the new and old changes which is not what I want since I am trying to keep the fragments separate. 

Am I missing a step here or is there a bug in the `menuconfig` step that is causing it to not pull the fragments into the config before it runs the tool?

Overall these kernel improvements are a big improvement though!

Thanks!

otavio, 2020-05-06:
You need to make multiple configuration fragments, for individual change/feature.

kevswims, 2020-05-07:
That's what I was trying to do. The issue I am having is that the changes from the first one are not reflected in menuconfig when I am working on the second one.

Somewhat unrelated to my initial issue. It's really difficult to make a slimmed down kernel based on the default defconfig because the default one contains so many selections that are not applicable to the hardware. It seems a little odd to me to have a bunch of configuration for SOCs that are not imx8's in the mx8 default defconfig. Would there be interest in working on a slimmed down one that only supports the hardware on the eval kit for example?

zandrey, 2020-05-07:
> That's what I was trying to do. The issue I am having is that the changes from the first one are not reflected in menuconfig when I am working on the second one.

Have you tried to generate first fragment, include it via `SRC_URI` and do a clean build? `menuconfig` together with `kernel_configme` is (from what I understand) brings the kernel to a known state which is the *defconfig* one.

This actually ensures that consecutive fragments would not include modifications introduced by already existing ones, hence the point from @otavio above is totally valid:

> You need to make multiple configuration fragments, for individual change/feature.

After every feature/change, include the new fragment and compile the kernel. This would ensure the iterative state of the modifications you're doing.

If you need to see one fragment feature from within another one then it means that you make fragments dependent on one another, and in this case it make sense to combine them into one, hence "working one feature at a time". 😉 

> 
> Somewhat unrelated to my initial issue. It's really difficult to make a slimmed down kernel based on the default defconfig because the default one contains so many selections that are not applicable to the hardware.

Agree, the *defconfig* has been taken from NXP tree, and it is indeed bloated with features unrelated to the *imx8* family (e.g. it contains nVidia drivers and ACPI support).

> It seems a little odd to me to have a bunch of configuration for SOCs that are not imx8's in the mx8 default defconfig.

I assume that NXP just took the *defconfig* from the *korg* tree and didn't sanitize it.

> Would there be interest in working on a slimmed down one that only supports the hardware on the eval kit for example?

This makes total sense, but brings another question: if NXP changes *defconfig* in their tree - how can we compare our sanitized version with the updated one?

-- andrey

otavio, 2020-05-07:
We can certainly work on a strimmed down defconfig but it needs to be done on our fork. And then we can send the patches to NXP.

kevswims, 2020-05-08:
@zandrey 

> Have you tried to generate first fragment, include it via SRC_URI and do a clean build? menuconfig together with kernel_configme is (from what I understand) brings the kernel to a known state which is the defconfig one.

This is exactly what I have tried to do. I have a config fragment that I created and added it to the `SRC_URI` in my BSP layer. I have verified that when I build the full kernel the configs set in it are correct. When I go back to make more changes I have run `do_clean`, `kernel_configme` and `menuconfig` it is not showing any of those config changes that I made in the config fragment. 

Kevin

closed 2021-02-17 by otavio