Freescale/meta-freescale #405: linux-imx-headers header files do not get added to Yocto SDKs

dv1, 2020-05-18:
I tried this with current master:

    MACHINE=imx8mmevk bitbake -c populate_sdk core-image-weston

The produced SDK did not contain the headers from linux-imx-headers, even though it is listed in the `RDEPENDS` list. I even tried to explicitely append linux-imx-headers to `TOOLCHAIN_TARGET_TASK`, and still, nothing.

zandrey, 2020-05-18:
@dv1 Do you see the same on `zeus` or only on `master`?

zandrey, 2020-05-18:
OK, I guess I've nailed it down.

I dropped `ALLOW_EMPTY_${PN} = "1"` in commit 824690b0cba597a943d883b72d26b968ff508ded, which caused all headers to land in `-dev` package. 🤦 

I'll have it tested and make a PR. Sorry for that.

-- andrey

zandrey, 2020-05-19:
It seems that the missing headers is more complex that I've imagined... 😞 I guess I would need a bit of advise here.

Adding `ALLOW_EMPTY_${PN} = "1"` does not solve the package generation, all headers are still landing in `-dev` package. Adding `BBCLASSEXTEND = "nativesdk"` also does not solve it.

@otavio Is it possible the `use-imx-headers` class does something which prevents those headers to be propagated to SDK? This is the only change I can think of right now...

@dv1 Can you please answer to my question above?

-- andrey

otavio, 2020-05-19:
I think you are looking the issue from the wrong perspective. I think you need to add a depedency to the dev of the package using the use-imx-headers so it adds the linux-imx-headers-dev so the image installs it if needed.

otavio, 2020-05-26:
This has been fixed on #409. Closing.

closed 2020-05-26 by otavio