Freescale/meta-freescale #438: cogl build is failing after upstream upgrade to 1.22.8

zandrey, 2020-06-18:
Upstream had `cogl` upgraded to *1.22.8*, and this causes a build failure with following errors:

```
 In file included from ../../cogl-1.22.8/cogl/cogl-egl-private.h:34,
                  from ../../cogl-1.22.8/cogl/winsys/cogl-winsys-private.h:48,
                  from ../../cogl-1.22.8/cogl/cogl-context-private.h:35,
                  from ../../cogl-1.22.8/cogl/cogl-gpu-info.c:41:
 ./cogl-egl-defines.h:38:10: fatal error: EGL/eglmesaext.h: No such file or directory
    38 | #include <EGL/eglmesaext.h>
       |          ^~~~~~~~~~~~~~~~~~
 compilation terminated.
```

Commit in `cogl` that causes this issue is [9c4764224a](https://gitlab.gnome.org/GNOME/cogl/-/commit/9c4764224aded552fb855b1c2b85b26d2b894adf), which explicitly includes `EGL/eglmesaext.h` header after migration to `libglvnd` is done. `EGL/eglmesaext.h` is non-upstreamed in `mesa`, and it is installed only when `egl` is enabled in `mesa` build.

This layer has EGL support in `mesa` disabled, which causes the installation of  `EGL/eglmesaext.h` header not to be performed, therefore causing the `cogl` build failure.

What I've also observed is that  `EGL/eglmesaext.h` header is present in 
```
cogl-1.0/1.22.8-r0/recipe-sysroot-native/usr/include/EGL/eglmesaext.h
```

while build uses `cogl-1.0/1.22.8-r0/recipe-sysroot` to search for includes files which obviously lacks the header.

If anyone has ideas on how to solve this - please share them here, as I was not able to find a clean way to resolve this build issue.

As a quick-n-dirty solution here, one can revert the above commit in `cogl` *1.22.8*, then build succeeds.

zandrey, 2020-06-27:
@petegriffin @otavio 

Looks like OE-Core upstream [has a patch](https://lists.openembedded.org/g/openembedded-core/message/140039) for this issue. Once it is merged in master - this issue can be closed as `cogl` should be buildable again.

I'll try to pick it up and test it with `meta-freescale` layer today.

-- andrey

otavio, 2020-06-27:
Awesome :-)

petegriffin, 2020-06-28:
@zandrey that is great news :)

zandrey, 2020-07-01:
As a follow-up here:

Commit 86172ed1cecd536a3dc6260622691db6cdf4d190 upstream has this build issue solved, and is present on the `master` branch.

I've just tested the build and for me it went through OK.

@petegriffin You might want to pull the updates `master` from *OE-Core* to verify it, and if you're AB shows no error - this issue can be closed.

-- andrey

petegriffin, 2020-07-02:
@zandrey I just kicked off another build to try and confirm, but I seem to be hitting some new issue now building the kernel for imx8mqevk https://ci.linaro.org/job/rpb-openembedded-master/1388/DISTRO=rpb-wayland,MACHINE=imx8mqevk,label=docker-stretch-amd64/consoleFull

13:06:33 NOTE: recipe fontconfig-native-2.13.1-r0: task do_fetch: Started
13:06:33 ERROR: linux-imx-5.4.3+gitAUTOINC+f8118585ee-r0 do_kernel_metadata: Could not locate BSP definition for imx8mqevk/standard and no defconfig was provided
13:06:33 ERROR: linux-imx-5.4.3+gitAUTOINC+f8118585ee-r0 do_kernel_metadata: Execution of '/srv/oe/build/tmp-rpb_wayland-glibc/work/imx8mqevk-linaro-linux/linux-imx/5.4.3+gitAUTOINC+f8118585ee-r0/temp/run.do_kernel_metadata.9306' failed with exit code 1:
13:06:33 WARNING: exit code 1 from a shell command.
13:06:33 
13:06:33 ERROR: Logfile of failure stored in: /srv/oe/build/tmp-rpb_wayland-glibc/work/imx8mqevk-linaro-linux/linux-imx/5.4.3+gitAUTOINC+f8118585ee-r0/temp/log.do_kernel_metadata.9306
13:06:33 Log data follows:

zandrey, 2020-07-02:
@petegriffin This is yet another issue, it is already addressed: https://lists.openembedded.org/g/openembedded-core/topic/75237251#140191

I've made a patch yesterday for this, and it is pending in upstream waiting for AB.

-- andrey

zandrey, 2020-07-19:
@petegriffin Ping, have you had time to verify this one is solved? Can this be closed now?

Thanks a lot!

-- andrey

petegriffin, 2020-07-21:
@zandrey yeah this is issue is fixed and can be closed :)  Also the kernel config issue you linked to above.

However I now see issues with gcc10 backported patch to optee-os when building optee-test. It is strange in that some of my builds are OK, but ta-dev-kit-dir isn't always defined

e.g. https://ci.linaro.org/job/rpb-openembedded-master/DISTRO=rpb-wayland,MACHINE=imx8mqevk,label=docker-stretch-amd64/lastBuild/console

14:54:00 | /srv/oe/build/tmp-rpb_wayland-glibc/work/cortexa53-crypto-linaro-linux/optee-test/3.2.0.imx-r0/recipe-sysroot/usr/include/optee/export-user_ta_arm64//mk/ta_dev_kit.mk:81: warning: undefined variable 'ta-dev-kit-dirta_arm64'
14:54:00 | /srv/oe/build/tmp-rpb_wayland-glibc/work/cortexa53-crypto-linaro-linux/optee-test/3.2.0.imx-r0/recipe-sysroot/usr/include/optee/export-user_ta_arm64//mk/ta_dev_kit.mk:81: warning: undefined variable 'COMPILER_ta_arm64'
14:54:00 | /srv/oe/build/tmp-rpb_wayland-glibc/work/cortexa53-crypto-linaro-linux/optee-test/3.2.0.imx-r0/recipe-sysroot/usr/include/optee/export-user_ta_arm64//mk/ta_dev_kit.mk:81: /mk/.mk: No such file or directory
14:54:00 | /srv/oe/build/tmp-rpb_wayland-glibc/work/cortexa53-crypto-linaro-linux/optee-test/3.2.0.imx-r0/recipe-sysroot/usr/include/optee/export-user_ta_arm64//mk/ta_dev_kit.mk:82: warning: undefined variable 'ta-dev-kit-dirta_arm64'
14:54:00 | /srv/oe/build/tmp-rpb_wayland-glibc/work/cortexa53-crypto-linaro-linux/optee-test/3.2.0.imx-r0/recipe-sysroot/usr/include/optee/export-user_ta_arm64//mk/ta_dev_kit.mk:82: /mk/cc-option.mk: No such file or directory

zandrey, 2020-08-03:
@petegriffin If that is OK for you - I would close this one down, so you can open another issue for the `optee-test` build failure.

I guess since `optee-test` does not relate to `cogl` - it should be tracked separately, otherwise people might get confused while searching for info under different header. 😉 

-- andrey

closed 2020-08-03 by zandrey