Freescale/meta-freescale #443: Python3-pycrypto-native missing

gmbankar37, 2020-07-07:
Hello, 
   I am trying to build an image with meta-freescale as a layer for i.mx8mmevk. But I am getting following error 

ERROR : Nothing PROVIDES ' python3-pycrypto-native' ( but meta-freescale/recipes-security/optee-imx/optee-os_3.2.0.imx.bb DEPENDS on or otherwise requires it )

I haved searched but cannot find the python3-pycrypto-native package. Can anyone help me on this issue? 
[image]

otavio, 2020-07-07:
What branches are you using?

gmbankar37, 2020-07-07:
I have cloned the meta-freescale layer from the git repository http://git.yoctoproject.org/cgit/cgit.cgi/meta-freescale . 
yocto project core dunfell 3.1.1 .

petegriffin, 2020-07-09:
In my Dunfell build it is located here meta-openembedded/meta-python/recipes-devtools/python/python3-pycrypto_2.6.1.bb

gmbankar37, 2020-07-21:
> In my Dunfell build it is located here meta-openembedded/meta-python/recipes-devtools/python/python3-pycrypto_2.6.1.bb

Thank you. This was useful. The meta-openembedded layer does not have a layer.conf file. I did create one but the error still exists. How did you include the meta-openembedded layer in your build?

IniterWorker, 2020-07-21:
@gmbankar37,

> Thank you. This was useful. The meta-openembedded layer does not have a layer.conf file. I did create one but the error still exists. How did you include the meta-openembedded layer in your build?

The meta-openembedded is a group of meta-layer. See the example below :

```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  ${HOME}/poky-dunfell/meta \
  ${HOME}/poky-dunfell/meta-poky \
  ${HOME}/poky-dunfell/meta-openembedded/meta-oe \
  ${HOME}/poky-dunfell/meta-openembedded/meta-networking \
  ${HOME}/poky-dunfell/meta-openembedded/meta-perl \
  ${HOME}/poky-dunfell/meta-openembedded/meta-python \
"
```

For more information, please see the documentation: [enabling-your-layer](https://www.yoctoproject.org/docs/2.1/dev-manual/dev-manual.html#enabling-your-layer).

B. r

gmbankar37, 2020-07-23:
Thank you @IniterWorker . The error is resolved and the image is getting build.

closed 2020-07-23 by gmbankar37