Freescale/meta-freescale #456: Integrate NXP Release 5.4.24-2.1.0

zandrey, 2020-08-17:
This is a collective issue to track the upgrade of community layer to NXP release `5.4.24-2.1.0` (based off `zeus` branch).

If anyone is currently working on upgrading any component or recipe - please post a recipe name (or name folder) with a short status here, so we can avoid double work here.

zandrey, 2020-08-17:
I'm currently on `recipes-bsp` part -> bumping up component versions to align them with NXP BSP.

As a side work (which should be done in order to support components in `recipe-bsp`) I'm also updating `conf/machine` files and `recipes-kernel/linux/linux-fslc-imx_5.4.bb` (with `defconfig`).

thochstein, 2020-08-24:
I'm working on linux-imx.

otavio, 2020-08-24:
@thochstein 
> I'm working on linux-imx.

Check the ongoing PR as @zandrey made our recipe, for our forked one.

thochstein, 2020-08-24:
@zandrey, did you have any issue with `CONFIG_CRYPTO_DES`?

```
WARNING: linux-imx-5.4.24+gitAUTOINC+babac008e5-r0 do_kernel_configcheck: [kernel config]: specified values did not make it into the kernel's final configuration:
...
    [INFO]: selection details for 'CONFIG_CRYPTO_DES':
        Symbols currently y-selecting this symbol:
          - CRYPTO_DEV_FSL_CAAM_CRYPTO_API_QI
          - CRYPTO_DEV_FSL_DPAA2_CAAM

        Symbols currently n-selecting this symbol (no effect):
          - INET_ESP
          - INET6_ESP
```

zandrey, 2020-08-24:
@thochstein I guess this comes from the fact that in `imx_v8_defconfig` is defined as `CONFIG_CRYPTO_DES=m` and at the beginning I've seen this popped up. I changed it to `CONFIG_CRYPTO_DES=y` which solved the warning.

Have a look at the `defconfig` in the ongoing PR, the only difference from `imx_v8_defconfig` it has is the `CONFIG_CRYPTO_DES`.

zandrey, 2020-08-31:
Just as a quick update here: I'm currently pacing further through recipes, following are locally upgraded:

```
imx-gpu-g2d: upgrade to 6.4.0.p2.4
imx-gpu-apitrace: fix the upgrade after rebase to 9.0.0
imx-dpu-g2d: upgrade to 1.8.7
imx-parser: upgrade to 4.5.5
imx-codec: update to 4.5.5
imx-vpuwrap: upgrade to 4.5.5
optee-imx: upgrade to version 3.7.0
firmware-imx: upgrade to version 8.8
```

I'd be looking at `gstreamer` next, and then come up with with PR to address Part 2 of the upgrade.

If anyone else is looking to help here - you're pretty much welcomed! 😉

thochstein, 2020-09-03:
Working on `imx-gpu-viv`.

thochstein, 2020-09-29:
Going through the machine conf, working on adding 6ULZ, 8DX, 8DXL so far...

zandrey, 2021-02-08:
@thochstein Are there any activities here that should be tracked in this issue, or can I close it?

I guess since the integration of `2.3.0` release in #622, this issue is not relevant anymore.

-- andrey

otavio, 2021-02-08:
Yes, it can be closed.

closed 2021-02-08 by otavio