Freescale/meta-freescale #465: perf is failing to build using linux-fslc

otavio, 2020-08-27:
Fail from our CI:

```
fsl-image-machine-test@imx53qsb | /srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/build/Makefile.build:144: recipe for target '/srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o' failed
fsl-image-machine-test@imx53qsb | make[3]: *** [/srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o] Error 1
fsl-image-machine-test@imx53qsb | Makefile.perf:597: recipe for target '/srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o' failed
fsl-image-machine-test@imx53qsb | make[2]: *** [/srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/perf-in.o] Error 2
fsl-image-machine-test@imx53qsb | Makefile.perf:220: recipe for target 'sub-make' failed
fsl-image-machine-test@imx53qsb | make[1]: *** [sub-make] Error 2
fsl-image-machine-test@imx53qsb | Makefile:69: recipe for target 'all' failed
fsl-image-machine-test@imx53qsb | make: *** [all] Error 2
fsl-image-machine-test@imx53qsb | make: Leaving directory '/srv/ci/nvme/work/imx53qsb-fslc-linux-gnueabi/perf/1.0-r9/perf-1.0/tools/perf'
fsl-image-machine-test@imx53qsb | WARNING: exit code 1 from a shell command.
fsl-image-machine-test@imx53qsb | 
fsl-image-machine-test@imx53qsb recipe perf-1.0-r9: task do_compile: Failed
fsl-image-machine-test@imx53qsb ERROR: Task (.../sources/poky/meta/recipes-kernel/perf/perf.bb:do_compile) failed with exit code '1'
fsl-image-machine-test@imx53qsb recipe babeltrace2-2.0.3-r0: task do_install: Succeeded
fsl-image-machine-test@imx53qsb recipe alsa-plugins-1.2.2-r0: task do_install: Succeeded
fsl-image-machine-test@imx53qsb recipe mpg123-1.26.3-r0: task do_configure: Succeeded
fsl-image-machine-test@imx53qsb recipe cairo-1.16.0-r0: task do_configure: Succeeded
fsl-image-machine-test@imx53qsb recipe mtd-utils-2.1.2-r0: task do_package_qa: Succeeded
fsl-image-machine-test@imx53qsb recipe mtd-utils-2.1.2-r0: task do_package_write_rpm: Succeeded
fsl-image-machine-test@imx53qsb recipe lttng-tools-2.12.2-r0: task do_configure: Succeeded
fsl-image-machine-test@imx53qsb recipe alsa-tools-1.2.2-r0: task do_configure: Succeeded
fsl-image-machine-test@imx53qsb recipe gdb-9.2-r0: task do_compile: Succeeded
fsl-image-machine-test@imx53qsb recipe ltp-20200515-r0: task do_compile: Succeeded
fsl-image-machine-test@imx53qsb recipe boost-1.74.0-r0: task do_install: Succeeded
fsl-image-machine-test@imx53qsb Tasks Summary: Attempted 5093 tasks of which 3924 didn't need to be rerun and 1 failed.
fsl-image-machine-test@imx53qsb Writing buildhistory
fsl-image-machine-test@imx53qsb Writing buildhistory took: 1 seconds
fsl-image-machine-test@imx53qsb 
fsl-image-machine-test@imx53qsb Summary: 1 task failed:
fsl-image-machine-test@imx53qsb   .../sources/poky/meta/recipes-kernel/perf/perf.bb:do_compile
fsl-image-machine-test@imx53qsb Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

zandrey, 2020-08-28:
@otavio Is it possible to get a full build log? I do not see the actual failure from the extract.

It might be that there are few more patches needed for `perf` to be cherry-picked, there has been some discussions on the oe-core about it.

BTW, I assume this is the latest `linux-fslc` with the `perf` patch applied, is that correct?

otavio, 2020-08-28:
No, it is before. Your patch fixes it, so please mention this issue in your PR so it is closed.

PR #467 (2020-08-28): Upgrade linux-fslc* kernel recipes with v5.4.61 + fix for perf build
Kernel branches are updated to *v5.4.61* from stable korg.

Update both `linux-fslc` and `linux-fslc-imx` recipes to include the latest patchlevel applied.

This update also includes the cherry-picked commit to address `perf` build issues for both kernels.

Fixes: #465 

-- andrey
closed 2020-08-28 by otavio