Freescale/meta-freescale #475: Disabled v4l feature in OpenCV

krzk, 2020-09-03:
Disabling V4L2 in dynamic-layers/openembedded-layer/recipes-support/opencv/opencv_4.%.bbappend is really a pain:
1. It's important feature of OpenCV.
2. Explanation by Otavio Salvador <otavio@ossystems.com.br> when adding this line is not even correct and you keep pushing it to all customers:
_"The camera support does not work using v4l. Disable it."_
This is simply not true, as V4L works fine and it is a proper way to get pictures from Linux mainline sensors.

Seriously... remove this patch.

otavio, 2020-09-03:
Hello @krzk, I checked the `bbappend` file and it disabled it only when using the NXP BSP as is quoted below:
```
FILESEXTRAPATHS_prepend := "${THISDIR}/${PN}:"

PACKAGECONFIG_remove_imxgpu2d = "v4l"
```

I am copying @thochstein here so he can ask internally if we can drop this override.

So I went further and checked where the comment you are referring to comes from and it seems you are referring to the following commit:
```
commit 1a344295d13fc5a483743037b99a996a47f38baa
Author: Otavio Salvador <otavio@ossystems.com.br>
Date:   Fri Jul 17 09:05:50 2015 -0300

    opencv: Remove v4l support for all i.MX6
    
    The camera support does not work using v4l. Disable it.
    
    Signed-off-by: Otavio Salvador <otavio@ossystems.com.br>
```

and as it can be seen on the commit log, this is from **2015** and at that time, we really were not using V4L. So it seems, you are referring to a quite old branch and I'd like to ask you to upgrade your layer revision.

krzk, 2020-09-03:
All NXP BSPs have imxgpu2d  even if they do not actually support it yet (e.g. i.MX 8).
Zeus branch is not old. You kept this change all the time, copying, updating, rebasing till OpenCV v4.3 and Yocto v3.0.

zandrey, 2020-09-04:
> All NXP BSPs have imxgpu2d even if they do not actually support it yet (e.g. i.MX 8).

That is not quite true, and according to [NXP Comparison table](https://www.nxp.com/docs/en/supporting-information/IMX8LAYERCMPR.pdf) *i.MX8M Nano* derivative lacks 2D HW accelerator or blit engine of some sort. I'm not sure what you mean by support in this case - packages or HW support?

Packages are present in this layer, as they are ported from NXP BSPs. HW 2D blit support can be cross-checked in the PDF link above.

> Zeus branch is not old. The change was kept all the time, copied, updated, rebased till OpenCV v4.3 and Yocto v3.0.

As for override - I guess @thochstein can shed some more light here.

krzk, 2020-09-07:
@zandrey I am saying that you're not even able to build the packages with meta-freescale. Just look at the compatible machines in recipes or their dependencies. Then also look at DTS which is missing important bits (although this is for running, not building). Anyway, this is an off-topic. We can keep argue on this, does not matter for the problem here.

There is no point in this override. The BSP layers should almost never disable other features just because at some time in the past they did not work perfectly. If they system crashes (which was not the case), could have sense. But policy "_oh, it does not work now, so I will disable it for everyone who includes meta-freescale (now and in the future)_" is not appropriate.

zandrey, 2020-09-07:
> @zandrey I am saying that you're not even able to build the packages with meta-freescale. 

Can you please be more specific about recipes that fails? Seriously, if there is a problem - this has to be addressed in the layer, but in order to do that a more detailed information regarding the particular error is needed.

Also, you might either extend this issue or open another one detailing your particular failure. I can tell you that my *OpenCV* build for example runs just fine.

> Just look at the compatible machines in recipes or their dependencies. Then also look at DTS which is missing important bits (although this is for running, not building). 

This is indeed interesting, care to share some details please?

> Anyway, this is an off-topic. We can keep argue on this, does not matter for the problem here.

Correct, let's move on to the problem.

> There is no point in this override. The BSP layers should almost never disable other features just because at some time in the past they did not work perfectly. If the system crashes (which was not the case), could have sense. But policy "_oh, it does not work now, so I will disable it for everyone who includes meta-freescale (now and in the future)_" is not appropriate.

Totally agree, therefore I've mentioned @thochstein here (and do it now again) so he can provide more details regarding why this append is present in this layer (and NXP BSP delivery as well).

I can only assume that this might be because:
1. Legacy reasons, which were solved but slipped through here
2. Lack of the feature support in NXP's `opencv` fork. In this case this should be addressed to NXP directly so they get to solve it

Don't get me wrong: I'm interested in this layer's stability, and findings like this one are targeted into exactly the direction of making things better. But a simple statement "_fix this because it's pain_" need to be further justified in terms of what "_pain_" is actually cause here. People might either not use `opencv` in their images and would never experience any issues, or they would face those issues and solve it themselves just because they are present upstream and not documented by some others who already found them.

-- andrey

zandrey, 2020-09-07:
As a follow-up:

Out of curiosity, I've tried to drop that *.bbappend* and `opencv` compiled successfully, together with `gstreamer1.0-plugins-bad` which pulls it as dependency.

I tend to lean towards point [1] from my above message, but NXP and @otavio gets the final say here.

-- andrey

otavio, 2020-09-07:
@zandrey 
> As a follow-up:
> 
> Out of curiosity, I've tried to drop that _.bbappend_ and `opencv` compiled successfully, together with `gstreamer1.0-plugins-bad` which pulls it as dependency.

Are you sure you did not use the mainline BSP? and did you use a machine that has a minimal set of GPU?

OpenCV is used (when using imx components) only at https://github.com/Freescale/meta-freescale/blob/master/recipes-multimedia/gstreamer/gstreamer1.0-plugins-bad_1.16.imx.bb#L6-L8

In case we use pristine GStreamer (https://github.com/Freescale/meta-freescale/blob/master/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.16.%25.bbappend) we don't enable it.

> I tend to lean towards point [1] from my above message, but NXP and @otavio gets the final say here.

If we confirm it builds I am fine in dropping it. We just need to be sure we don't break builds.

For this, the easier is to use `fsl-wayland` distro check if `opencl` and `gst` builds for example for `imx6slevk`.

zandrey, 2020-09-07:
> Are you sure you did not use the mainline BSP? and did you use a machine that has a minimal set of GPU?

I used the NXP BSP  and just built *imx8mnevk* machine (no VPU, no 2D HW support) and with *opencv.bbappend* dropped - it went through.

krzk, 2020-09-07:
It seems you connect disabling of OpenCV V4L2 with building/running gstreamer imx plugins. This is fundamentally wrong approach because:
1. Anyone could want to run OpenCV V4L2 without any iMX GST plugins at all,
2. There is no single reason to disallow such configuration for all customers.

Of course, you might say "you do not need V4L2, use NXP camera stack"... but then you forgot that NXP V4L2 stack for i.MX 8M is really bad (far away from uptreamable quality, not supporting mainline sensors except highly customized OV56xx) and does not work on Mini (no ISP, no OpenCL). With effort (took me some time) is fixable and I have now somehow working iMX camera stack with mainline sensors. This means - mainline v4l2 interface with subdevices and controls, not these terribly hacked OV56xx with three different versions of same driver (ov5640.c, ov5640_mipi.c, ov5640_mipi_v2.c, ov5640_v2.c...).

I really do not get why we keep discussing about this disabled v4l2... It works fine, compiles and if regular, vanilla OpenCV package breaks any NXP Gstreamer plugins, the plugins are the problem. Not the OpenCV.

otavio, 2020-09-07:
@krzk let me be very straight with you, please stop complaining and send a PR showing what you are proposing. 

We're very open to contributions and we'd love to receive a PR with detailed commit logs for the changes showing what you intend to do and as well as how you tested this. Bear in mind we're supporting all SoCs and not only 8 families so we need to be sure to consider the other SoCs. If you provide a PR, it is easier for us to reason about the change you're proposing.

So please, prepare the PR and we can focus on the code changes you're proposing.

krzk, 2020-09-07:
Sure :)

thochstein, 2020-09-08:
Hi folks, sorry for my late response. I'm checking with the team now.

thochstein, 2020-09-10:
It appears that the dependency should have been removed some years ago...

otavio, 2020-09-10:
Mind to prepare a PR?

thochstein, 2020-09-10:
Okay: https://github.com/Freescale/meta-freescale/pull/486

PR #488 (2020-09-10): opencv: Drop removal of v4l support
In the past, i.MX gstreamer camera didn't work well with OpenCV v4l,
so v4l was disabled. Now that i.MX gstreamer is no longer using
OpenCV, v4l support can be restored.

Fixes #475
commit cf028de9 (2020-09-10): opencv: Drop removal of v4l support

In the past, i.MX gstreamer camera didn't work well with OpenCV v4l,
so v4l was disabled. Now that i.MX gstreamer is no longer using
OpenCV, v4l support can be restored.

Fixes #475
commit 30234f74 (2020-09-14): opencv: Drop removal of v4l support

In the past, i.MX gstreamer camera didn't work well with OpenCV v4l,
so v4l was disabled. Now that i.MX gstreamer is no longer using
OpenCV, v4l support can be restored.

Fixes #475
commit 69330ea2 (2020-09-14): opencv: Drop removal of v4l support

In the past, i.MX gstreamer camera didn't work well with OpenCV v4l,
so v4l was disabled. Now that i.MX gstreamer is no longer using
OpenCV, v4l support can be restored.

Fixes #475

(cherry picked from commit 30234f745c55e5f832c282d51bb44604f151e909)
closed 2020-09-14 by otavio