Freescale/meta-freescale #476: Is it possible to use HAB with meta-freescale?

RobertBerger, 2020-09-04:
If so, is this documented somewhere?

otavio, 2020-09-04:
It is, but it is not documented anywhere.

zandrey, 2020-09-04:
@otavio Is there *any* piece of information available on this?

People are asking about this in Yocto IRC a lot, but there is no place to adress them to. If you have anything regarding this - can you please share it?

Thanks a lot!

-- andrey

zandrey, 2020-09-07:
@RobertBerger 

I just found that there is a documentation for HAB in U-Boot source tree: https://github.com/Freescale/u-boot-fslc/blob/e2cb327cff2d12d715d89c8779a0fdfcfc39c1e9/doc/imx/habv4/introduction_habv4.txt

This was mentioned by *qschulz* on yocto IRC, and I checked the `u-boot-fslc` source tree if the documentation in there - the description from DENX is present there.

RobertBerger, 2020-09-08:
Cool thanks! So no need to sign an NDA with Freescale ;)

ricardosalveti, 2020-09-08:
The HAB stuff you can see without an NDA (I believe NDAs are only required for a few socs so you know which fuses to blow).  We have some docs for our own yocto-based distro (which uses meta-freescale) at https://docs.foundries.io/latest/reference/secure-boot.html, in case you want to have a look how we're doing secure boot.

angolini, 2021-04-13:
I think this issue can be closed now.

In case anyone from the future has anything else to add, please fell free to reopen it.

closed 2021-04-13 by angolini

zandrey, 2021-06-10:
Just as an update to HAB topic:

There was an [announcement on the U-Boot ML](https://lore.kernel.org/u-boot/7a08f509-c216-9d8e-81a6-6a26079011ef@denx.de/):

_DENX_ published the layer which takes care about HAB on the i.MX8M series and have it working with upstream components, so anyone interested in integrating HAB in their products might have a look at it as a potential solution.

Link to repository: https://source.denx.de/denx/meta-secure-imx

-- andrey