Freescale/meta-freescale #484: problem building linux-fslc-imx on dunfell

cpboyd, 2020-09-09:
With the recent commit changes, I now get this issue building the kernel even after removing my own `bbappend` files:
```
/home/cpboyd/9300-yocto/build_fb/tmp/work-shared/imx6qdl-iwg15-sm/kernel-source/drivers/media/platform/mxc/output/mxc_vout.c: In function 'config_disp_output':
/home/cpboyd/9300-yocto/build_fb/tmp/work-shared/imx6qdl-iwg15-sm/kernel-source/drivers/media/platform/mxc/output/mxc_vout.c:1873:16: error: 'FBINFO_MISC_USEREVENT' undeclared (first use in this function)
 1873 |  fbi->flags |= FBINFO_MISC_USEREVENT;
      |                ^~~~~~~~~~~~~~~~~~~~~
/home/cpboyd/9300-yocto/build_fb/tmp/work-shared/imx6qdl-iwg15-sm/kernel-source/drivers/media/platform/mxc/output/mxc_vout.c:1873:16: note: each undeclared identifier is reported only once for each function it appears in
  AR      net/sunrpc/built-in.a
  CC      drivers/gpu/drm/drm_hashtab.o
/home/cpboyd/9300-yocto/build_fb/tmp/work-shared/imx6qdl-iwg15-sm/kernel-source/drivers/media/platform/mxc/output/mxc_vout.c: In function 'release_disp_output':
/home/cpboyd/9300-yocto/build_fb/tmp/work-shared/imx6qdl-iwg15-sm/kernel-source/drivers/media/platform/mxc/output/mxc_vout.c:1995:16: error: 'FBINFO_MISC_USEREVENT' undeclared (first use in this function)
 1995 |  fbi->flags |= FBINFO_MISC_USEREVENT;
      |                ^~~~~~~~~~~~~~~~~~~~~
```

cpboyd, 2020-09-09:
This commit removed the relevant `#define` from `include/linux/fb.h`:
https://github.com/Freescale/linux-fslc/commit/49fea9ddec254c85351bf1417da1730ede6f8a2e#diff-43768cc89c01c730cb119634b42dfa81

otavio, 2020-09-09:
@zandrey can you take a look?

@cpboyd what `defconfig` are you using?

cpboyd, 2020-09-09:
@otavio One based off of `imx_v7_defconfig`

It seems like anything including `CONFIG_VIDEO_MXC_IPU_OUTPUT=y` should result in this error.

zandrey, 2020-09-10:
@cpboyd `FBINFO_MISC_USEREVENT` has been removed in upstream with commit [376810e5e9e148648c8b6e4aa8dc6306e9364f49](https://github.com/Freescale/linux-fslc/commit/376810e5e9e148648c8b6e4aa8dc6306e9364f49).

I'd say that *drivers/media/platform/mxc/output/mxc_vout.c* would require same changes, since it is not present in upstream.

If you can port that patch and send it over against `linux-fslc` tree - that would be great, otherwise I can take over it but that might take some time, probably till the beginning of next week.

-- andrey

zandrey, 2020-09-10:
@cpboyd I've come up with a patch in my fork for this: [bffbf24c88a75f0e0979fbd6eca034e010230e57](https://github.com/zandrey/linux-fslc/commit/bffbf24c88a75f0e0979fbd6eca034e010230e57)

Can you pick it up and test it? If that passes for you - I would open a PR to have it in `5.4-1.0.0-imx`.

-- andrey

cpboyd, 2020-09-11:
@zandrey The build compiled successfully. I haven't actually tested it yet

zandrey, 2020-09-14:
@cpboyd Thanks a lot for a quick feedback!

Is it possible you test this patch in runtime? I do not have a HW to verify it, so I would need to rely on your further feedback here.

Once you can verify it is operable - I would include it also into next kernel update, together with *v5.4.65* merge.

-- andrey

cpboyd, 2020-09-14:
@zandrey It seems that my build has issues related to `CONFIG_OF_OVERLAY` that I'm now looking into

zandrey, 2020-09-14:
@cpboyd Thanks for testing it out!

I'd queue up this patch then for next kernel update.

Cheers!
-- andrey

PR #494 (2020-09-15): Consolidated kernel update (v5.4.65)
Both kernel branches were updated to *v5.4.65* from stable korg, recipe `SRCREV` are updated to point to that version now.

`linux-fslc-imx` recipe contains additional fix to address the build issue due to `FBINFO_MISC_USEREVENT` define has been dropped upstream.

Fixes: #484 

@thochstein Please forward the patch [bffbf24c88a75f0e0979fbd6eca034e010230e57](https://github.com/Freescale/linux-fslc/commit/bffbf24c88a75f0e0979fbd6eca034e010230e57) to your kernel team, it would be required once you would advance past *v5.4.62* from upstream.

-- andrey
closed 2020-09-15 by otavio