Freescale/meta-freescale #508: Compile Error in uboot2020.04 in current dunfell branch

gharti, 2020-10-02:
In current dunfell branch u-boot-imx_2020.04.bb does not compile

Error message:
```
In file included from /home/devel/src/bsp-dunfell/build-fsl-wayland/tmp/work/imx6qdlsabresd-fsl-linux-gnueabi/u-boot-imx/2020.04-r0/git/board/freescale/mx6sabresd/mx6sabresd.c:36:
include/asm/arch/mx6-ddr.h:24:2: error: #error "Please select cpu"
   24 | #error "Please select cpu"
		  |  ^~~~~
```	  

And a lot of warnings about redefind constants like
```
mx6sabre_common.h:402: warning: "CONFIG_CMD_BMP" redefined
  402 | #define CONFIG_CMD_BMP
		  | 
In file included from /home/devel/src/bsp-dunfell/build-fsl-wayland/tmp/work/imx6qdlsabresd-fsl-linux-gnueabi/u-boot-imx/2020.04-r0/git/include/linux/kconfig.h:4,
					 from <command-line>:
include/generated/autoconf.h:122: note: this is the location of the previous definition
  122 | #define CONFIG_CMD_BMP 1
```

This happens in a clean checkout and build, i.e:
```
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b dunfell
repo sync
MACHINE=imx6qdlsabresd DISTRO=fsl-wayland source setup-environment build-fsl-wayland
bitbake -k fsl-image-machine-test
```

otavio, 2020-10-02:
@thochstein can you take a look at this?

thochstein, 2021-02-18:
I'm not able to reproduce the compile error. @gharti can you make sure you have updated because it seems that your error was fixed by Max.

Unfortunately, I am seeing an install error on both `dunfell` and `master`:

```
| install: cannot stat '/opt/work/upstream/fsl-xwayland/tmp/work/imx6qdlsabresd-fsl-linux-gnueabi/u-boot-imx/2020.04-r0/build/u-boot.img': No such file or directory
```

That suffix looks incorrect, seems it should be `imx` for `fsl-wayland`, but the MACHINE setting seems to be overriding even a change in `local.conf`. It doesn't explain the install error, though. Will have to look more tomorrow.

thochstein, 2021-02-19:
@otavio I think we're running into design problems due to the duality of mainline and NXP BSP and the moving of some of that design into the distro config. @zandrey updated the distro config to fix a problem with the 8M config, but that creates this problem here for imx6qdlsabresd which is defaulting to mainline bsp. I don't have a good grasp of the big picture here, so don't know how to fix it. I'll keep looking at it but wanted to mention it in case it's more obvious to you or Andrey.

thochstein, 2021-02-19:
I'm about to test this, but my idea is to undo the commit by @zandrey and update the commit by @MaxKrummenacher so the `SPL_BINARY` overrides are only for 6 and 7.

zandrey, 2021-02-19:
> 
> 
> I'm about to test this, but my idea is to undo the commit by @zandrey and update the commit by @MaxKrummenacher so the `SPL_BINARY` overrides are only for 6 and 7.

@thochstein I'm not sure which commit of mine you're referring to. Is it the [880dbf2951](https://github.com/Freescale/meta-freescale-distro/commit/880dbf2951e4fad5985825a225f26acb2ddb6041) in _meta-freescale-distro_ repository or some other one?

-- andrey

thochstein, 2021-02-19:
> @thochstein I'm not sure which commit of mine you're referring to. Is it the [880dbf2951](https://github.com/Freescale/meta-freescale-distro/commit/880dbf2951e4fad5985825a225f26acb2ddb6041) in _meta-freescale-distro_ repository or some other one?

Yes, that's the one. Do you think it will be okay? I've built `imx6qdlsabresd` and `imx8qm-mek`, waiting for `imx8mm-lpddr4-evk`.

gharti, 2021-02-22:
@thochstein: With current checkout, the build also works on my side. Thanks. 
But now I also see the same install error as you mentioned.

otavio, 2021-03-19:
@gharti can you check it now? if it is reproducable, please let us know.

closed 2021-03-24 by otavio