Freescale/meta-freescale #509: Configuration error in mesa package for the imx6qdlsabresd reference platform

gharti, 2020-10-02:
In current dunfell branch there is a bitbake configure error in mesa package

Error in sources/poky/meta/recipes-graphics/mesa/mesa_20.0.2.bb:do_configure

Error message
```
meson.build:455:4: ERROR: Problem encountered: building dri drivers require at least one windowing system or classic osmesa
```

This happens in a clean checkout and build, i.e:
```
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b dunfell
repo sync
MACHINE=imx6qdlsabresd DISTRO=fsl-wayland source setup-environment build-fsl-wayland
bitbake -k fsl-image-machine-test
```

otavio, 2020-10-02:
Could you check it against `fslc-wayland`?

@thochstein can you check mesa at your side?

zandrey, 2020-10-02:
IIRC, we used to have similar issue (#115), which was finally solved by @fschrempf with his PR #287 

@gharti Can you take a look at the PR to see if you can apply similar mechanisms for `imx6qdlsabresd`?

-- andrey

gharti, 2020-10-05:
It is possible to add a USE_OSMESA_ONLY_mx6 ?= "yes" to a mesa.bbappend file like it is done for the i.MX8, but then swrast is used for DRI.

But is this the best way to do it on i.MX6 because swrast is used then? 

`fsl-wayland` with `USE_OSMESA_ONLY_mx6`
```
$bitbake -e mesa | grep ^DRI
DRIDRIVERS="swrast"
```

Plain `fslc-wayland` and plain `fsl-wayland` (and also NXP Zeus release):
```
$bitbake -e mesa | grep ^DRI
DRIDRIVERS=""
```

But `fslc-wayland` uses ETNAVIV while `fsl-wayland` uses VIVANTE driver. Isn't it possible to get VIVANTE driver to work on fsl-wayland like on Zeus for iMX6?

otavio, 2020-10-05:
You can use it with etnaviv if you add:

```
IMX_DEFAULT_BSP = "nxp"
```

at your conf/local.conf, so it overrides it for `fslc-wayland`.

gharti, 2020-10-07:
I tried this approach today with the result that the mesa configuration error also occure.

otavio, 2021-02-01:
@thochstein can you validate this?

otavio, 2021-02-17:
Fixed in  #634

closed 2021-02-17 by otavio