Freescale/meta-freescale #510: Compile error in lttng-modules for imx6qdlsabresd on dunfell (and dunfell-next)

gharti, 2020-10-05:
An error occured on `sources/poky/meta/recipes-kernel/lttng/lttng-modules_2.11.2.bb:do_compile`:

```
/home/devel/src/bsp-dunfell-fslc/build-fslc-wayland/tmp/work/imx6qdlsabresd-fslc-linux-gnueabi/lttng-modules/2.11.2-r0/lttng-modules-2.11.2/probes/../probes/lttng-tracepoint-event-impl.h:130:6: error: conflicting types for 'trace_writeback_queue_io'
  130 | void trace_##_name(_proto);
      |      ^~~~~~

```

This happens in a clean checkout and build, i.e:
```
repo init -u https://github.com/Freescale/fsl-community-bsp-platform -b dunfell
repo sync
MACHINE=imx6qdlsabresd DISTRO=fslc-wayland source setup-environment build-fsl-wayland
bitbake -k fsl-image-machine-test
```

And also if switched poky to dunfell-next

otavio, 2020-10-05:
The PR is here:

http://cgit.openembedded.org/openembedded-core-contrib/log/?h=stable/dunfell-next; please apply this (http://cgit.openembedded.org/openembedded-core-contrib/patch/?id=348c666b2dca230308c8462dac2117b04cca6ae7) and let us know if it works for you.

gharti, 2020-10-06:
Hi Otavio, dunfell-next with this patch applied works. Thanks. Will it be backported to dunfell?

otavio, 2020-10-06:
Yes; it is queued for merge. I am closing this issue as we cannot fix it here.

closed 2020-10-06 by otavio