Freescale/meta-freescale #518: Force pushing branches?

ernstp, 2020-10-09:
I'm getting a number of these from meta-freescale... shouldn't be necessary?
 ! [rejected]          dunfell -> dunfell (non-fast-forward)
 ! [rejected]          master -> master (fetch first)

otavio, 2020-10-09:
Hello,

Em sex., 9 de out. de 2020 às 11:03, Ernst Sjöstrand <
notifications@github.com> escreveu:

> I'm getting a number of these from meta-freescale... shouldn't be
> necessary?
> ! [rejected] dunfell -> dunfell (non-fast-forward)
> ! [rejected] master -> master (fetch first)
>

I made a force push some time ago but I don't recall doing it lately. Maybe
you did not update it recently?

-- 
Otavio Salvador                             O.S. Systems
http://www.ossystems.com.br        http://code.ossystems.com.br
Mobile: +55 (53) 9 9981-7854          Mobile: +1 (347) 903-9750

ernstp, 2020-10-09:
It was from an automated system that I haven't checked in a while. Anyway, good if it can be avoided!

closed 2020-10-09 by ernstp