Freescale/meta-freescale #521: Invalid version numbers in linux-fslc-imx and kernel-module-imx-gpu-viv

dv1, 2020-10-15:
These recipes have version numbers like "-imx_5.4.24_2.1.0". However, underscores are not permitted in version numbers.

Is it safe to simply replace the underscores with dashes?

otavio, 2020-10-15:
Which package manager?

dv1, 2020-10-15:
ipk .

zandrey, 2020-10-15:
Interesting, I've never caught this issue on me end...

Nevertheless, @dv1 is right - those symbols are not valid in version number.

I'm doing a merge of _5.4.71_ now and can queue this change also up, if there are no objections here. Would opt in for `-imx-5.4.24-2.1.0`, it would more-or-less match the kernel branch used by NXP.

-- andrey

otavio, 2020-10-15:
Sure, please prepare it and send it with the PR :-)

zandrey, 2020-10-16:
@dv1 I was trying to reproduce the issue you have with versioning, but so far I could not find any combination of `DISTRO_FEATURES` and machine to have this error pop up.

I've actually changed the versioning scheme in the `linux-fslc-imx` recipe to `-imx-5.4.24-2.1.0` as it should be (PR follows shortly), but it would be good to know what was the combination you used that triggered this behavior.

-- andrey

zandrey, 2020-10-16:
> I believe he was using `ipk` packaging.

I have packages also configured as `PACKAGE_CLASSES = "package_ipk"` in _local.conf_ and still - I do not see the error... This is what really puzzled me and that is why I wanted to get a bit more info on the build config @dv1 is using to reproduce the issue.

dv1, 2020-10-16:
I use current git master, and poky.

These are the layers:

    poky/meta
    meta-freescale
    meta-freescale-3rdparty
    meta-raspberrypi
    meta-qt5
    meta-openembedded/meta-multimedia
    meta-openembedded/meta-networking
    meta-openembedded/meta-gnome
    meta-openembedded/meta-python
    meta-openembedded/meta-oe
    poky/meta-poky
    poky/meta-yocto-bsp

And this is from the local.conf:

    LICENSE_FLAGS_WHITELIST = "commercial"
    PACKAGE_CLASSES = "package_ipk"
    INHERIT += "rm_work"
    RM_WORK_EXCLUDE += "virtual/kernel"
    ACCEPT_FSL_EULA = "1"

    EXTRA_IMAGE_FEATURES += "ssh-server-dropbear package-management ptest-pkgs"

    TOOLCHAIN_HOST_TASK_append = " nativesdk-meson "

    IMX_DEFAULT_BSP = "nxp"
    IMAGE_FSTYPES = "wic.gz tar.gz"

    SDK_XZ_COMPRESSION_LEVEL = "-0"

PR #522 (2020-10-16): Consolidated Kernel update (v5.4.71)
Both kernel branches were updated to _v5.4.71_ from stable korg, recipe `SRCREV` are updated to point to that version now.

`linux-fslc` received a separate patch to address VPU FW loading during probing.

`linux-fslc-imx` has updated versioning scheme recorded in `LOCALVERSION` to address build errors due to incompatible symbols were used (Fixes: #521).

-- andrey
closed 2020-10-16 by otavio