Freescale/meta-freescale #528: [IMX6Q][Dunfell] Problem using imxeglvivsink on top of X11

woutervanh, 2020-10-20:
I'm using a custom board/custom distro, based on yocto Dunfell. It's using vivante drivers (so not etnaviv).
Running kernel is: 5.4-2.1.x-imx
X11 itself is working fine. Although I get a VivDRIScreenInit -releated error message in the log:

```
X.Org X Server 1.20.8
X Protocol Version 11, Revision 0
[    34.528] Build Operating System: Linux 
[    34.528] Current Operating System: Linux imx6qtelevic_generic 5.4.46-04202-gd069a637b280 #15 SMP PREEMPT Wed Oct 14 12:13:42 CEST 2020 armv7l
[    34.528] Kernel command line: version=DEV_33968064_1-04-02 console=ttymxc3,115200 ubi.mtd=0 root=ubi0:rootfsA rootfstype=ubifs mtdparts=gpmi-nand:-(ubipart) quiet
[    34.528] Build Date: 29 March 2020  08:21:46PM
[    34.528]  
[    34.534] Current version of pixman: 0.38.4
[    34.535] Before reporting problems, check http://wiki.x.org
to make sure that you have the latest version.
[    34.535] Markers: (--) probed, (**) from config file, (==) default setting,
(++) from command line, (!!) notice, (II) informational,
(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    34.538] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 12:41:08 2018
[    34.565] (==) Using config file: "/etc/X11/xorg.conf"
[    34.566] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    34.574] (==) No Layout section.  Using the first Screen section.
[    34.576] (==) No screen section available. Using defaults.
[    34.576] (**) |-->Screen "Default Screen Section" (0)
[    34.576] (**) |   |-->Monitor "<default monitor>"
[    34.584] (==) No device specified for screen "Default Screen Section".
Using the first device section listed.
[    34.584] (**) |   |-->Device "i.MX Accelerated Framebuffer Device"
[    34.584] (==) No monitor specified for screen "Default Screen Section".
Using a default monitor configuration.
[    34.585] (**) Option "BlankTime" "0"
[    34.585] (**) Option "StandbyTime" "0"
[    34.585] (**) Option "SuspendTime" "0"
[    34.585] (**) Option "OffTime" "0"
[    34.586] (==) Automatically adding devices
[    34.586] (==) Automatically enabling devices
[    34.586] (==) Automatically adding GPU devices
[    34.588] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    34.589] (WW) The directory "/usr/share/fonts/X11/misc/" does not exist.
[    34.589] Entry deleted from font path.
[    34.590] (WW) The directory "/usr/share/fonts/X11/TTF/" does not exist.
[    34.590] Entry deleted from font path.
[    34.590] (WW) The directory "/usr/share/fonts/X11/OTF/" does not exist.
[    34.590] Entry deleted from font path.
[    34.590] (WW) The directory "/usr/share/fonts/X11/Type1/" does not exist.
[    34.590] Entry deleted from font path.
[    34.590] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    34.590] Entry deleted from font path.
[    34.590] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    34.590] Entry deleted from font path.
[    34.590] (==) FontPath set to:
 
[    34.591] (==) ModulePath set to "/usr/lib/xorg/modules"
[    34.591] (II) The server relies on udev to provide the list of input devices.
If no devices become available, reconfigure udev or disable AutoAddDevices.
[    34.600] (II) Loader magic: 0x5f8d80
[    34.600] (II) Module ABI versions:
[    34.600] X.Org ANSI C Emulation: 0.4
[    34.601] X.Org Video Driver: 24.1
[    34.601] X.Org XInput driver : 24.1
[    34.601] X.Org Server Extension : 10.0
[    34.610] (II) no primary bus or device found
[    34.614] (II) LoadModule: "glx"
[    34.630] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    34.793] (II) Module glx: vendor="X.Org Foundation"
[    34.794] compiled for 1.20.8, module version = 1.0.0
[    34.794] ABI class: X.Org Server Extension, version 10.0
[    34.797] (II) LoadModule: "vivante"
[    34.798] (II) Loading /usr/lib/xorg/modules/drivers/vivante_drv.so
[    34.809] (II) Module vivante: vendor="X.Org Foundation"
[... 125 lines cut ...]
DISTRO ?= "fslc-x11"
ACCEPT_FSL_EULA = "1"
IMX_DEFAULT_BSP = "nxp"
CORE_IMAGE_EXTRA_INSTALL += " \
    packagegroup-fsl-gstreamer1.0-full \
"
```
With none of my own layers enabled, and the problem persists.

```
root@imx6qdlsabresd:~# uname -an
Linux imx6qdlsabresd 5.4.70+g2d1241e38347 #1 SMP PREEMPT Wed Oct 7 17:10:48 UTC 2020 armv7l GNU/Linux
root@imx6qdlsabresd:~# DISPLAY=:0 gst-launch-1.0 videotestsrc ! imxeglvivsink
Setting pipeline to PAUSED ...
Xlib:  extension "XFree86-DRI" missing on display ":0".
Pipeline is PREROLLING ...
Xlib:  extension "XFree86-DRI" missing on display ":0".
Xlib:  extension "XFree86-DRI" missing on display ":0".
Pipeline is PREROLLED ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 474' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

otavio, 2020-10-20:
@woutervanh thanks for reporting it here; I'm wondering if you could get an image with debugging symbols and see if `gdb` could shed any light on the culprit?

MaxKrummenacher, 2020-10-20:
Hi @woutervanh

I observed similar symptoms when using X11 with i.MX 6. i.e. the 'Xlib:  extension "XFree86-DRI" missing on display ":0"'

That one I could overcome by [forward porting](http://git.toradex.com/cgit/linux-toradex.git/commit/drivers?h=toradex_5.4-2.1.x-imx&id=c0ad2a24a02344e50109f436fd296e79bd740e0c) the DRI1 Vivante driver in the kernel. Looks like DRI1 got deprecated with a nice warning about opening a security hole when using DRI1. This probably made NXP not forward port it.

While with that driver enabled the '"XFree86-DRI" missing' message no longer printed any application which uses the GPU now segfaults. I trieded with glmark2 and I guess also with the EGL sink as you do.

I had to put the issue aside at the moment. Thought I could report what I've seen though.

woutervanh, 2020-10-20:
> @woutervanh thanks for reporting it here; I'm wondering if you could get an image with debugging symbols and see if `gdb` could shed any light on the culprit?

Maybe I can try tomorrow, but I'm not familiar with gdb...

woutervanh, 2020-10-21:
> Hi @woutervanh
>
> I observed similar symptoms when using X11 with i.MX 6. i.e. the 'Xlib: extension "XFree86-DRI" missing on display ":0"'
>
> That one I could overcome by [forward [...]

I think you saved my day! 
I also took your patch and enabled DRM_VIVANTE in the kernel. Warnings are gone and I don't get a segfault any longer. I'll have to verify if I have an actual visual when I have display at hand (which I don't have right now) but it looks good.
Did not try glmark2 though

otavio, 2020-10-21:
Oh! cool! @woutervanh, looking forward to your test results!

otavio, 2020-10-21:
@psivasubramanian @thochstein is this a known issue on NXP releases?

woutervanh, 2020-10-21:
```
root@imx6q_generic:~# DISPLAY=:0 glmark2     
Error: Failed to load libGL
Segmentation fault
```

```
root@imx6q_generic:~# DISPLAY=:0 glmark2-es2 
=======================================================
    glmark2 2017.07
=======================================================
    OpenGL Information
    GL_VENDOR:     Vivante Corporation
    GL_RENDERER:   Vivante GC2000
    GL_VERSION:    OpenGL ES 3.0 V6.4.0.p2.234062
=======================================================
[build] use-vbo=false: FPS: 116 FrameTime: 8.621 ms
[build] use-vbo=true: FPS: 144 FrameTime: 6.944 ms
[texture] texture-filter=nearest: FPS: 130 FrameTime: 7.692 ms
[texture] texture-filter=linear: FPS: 127 FrameTime: 7.874 ms
[texture] texture-filter=mipmap: FPS: 124 FrameTime: 8.065 ms
[shading] shading=gouraud: FPS: 134 FrameTime: 7.463 ms
[shading] shading=blinn-phong-inf: FPS: 132 FrameTime: 7.576 ms
[shading] shading=phong: FPS: 127 FrameTime: 7.874 ms
[shading] shading=cel: FPS: 121 FrameTime: 8.264 ms
[bump] bump-render=high-poly: FPS: 129 FrameTime: 7.752 ms
[bump] bump-render=normals: FPS: 144 FrameTime: 6.944 ms
[bump] bump-render=height: FPS: 140 FrameTime: 7.143 ms
[effect2d] kernel=0,1,0;1,-4,1;0,1,0;: FPS: 94 FrameTime: 10.638 ms
[effect2d] kernel=1,1,1,1,1;1,1,1,1,1;1,1,1,1,1;: FPS: 61 FrameTime: 16.393 ms
[pulsar] light=false:quads=5:texture=false: FPS: 108 FrameTime: 9.259 ms
[desktop] blur-radius=5:effect=blur:passes=1:separable=true:windows=4: FPS: 45 FrameTime: 22.222 ms
[desktop] effect=shadow:windows=4: FPS: 82 FrameTime: 12.195 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 56 FrameTime: 17.857 ms
[buffer] columns=200:interleave=false:update-dispersion=0.9:update-fraction=0.5:update-method=subdata: FPS: 57 FrameTime: 17.544 ms
[buffer] columns=200:interleave=true:update-dispersion=0.9:update-fraction=0.5:update-method=map: FPS: 69 FrameTime: 14.493 ms
[ideas] speed=duration: FPS: 77 FrameTime: 12.987 ms
[jellyfish] <default>: FPS: 86 FrameTime: 11.628 ms
[terrain] <default>: FPS: 0 FrameTime: inf ms
[shadow] <default>: FPS: 83 FrameTime: 12.048 ms
[refract] <default>: FPS: 22 FrameTime: 45.455 ms
[conditionals] fragment-steps=0:vertex-steps=0: FPS: 141 FrameTime: 7.092 ms
[conditionals] fragment-steps=5:vertex-steps=0: FPS: 122 FrameTime: 8.197 ms
[conditionals] fragment-steps=0:vertex-steps=5: FPS: 140 FrameTime: 7.143 ms
[function] fragment-complexity=low:fragment-steps=5: FPS: 130 FrameTime: 7.692 ms
[function] fragment-complexity=medium:fragment-steps=5: FPS: 122 FrameTime: 8.197 ms
[loop] fragment-loop=false:fragment-steps=5:vertex-steps=5: FPS: 130 FrameTime: 7.692 ms
[loop] fragment-steps=5:fragment-uniform=false:vertex-steps=5: FPS: 129 FrameTime: 7.752 ms
[loop] fragment-steps=5:fragment-uniform=true:vertex-steps=5: FPS: 122 FrameTime: 8.197 ms
=======================================================
                                  glmark2 Score: 104 
=======================================================

```

otavio, 2020-10-22:
@woutervanh could you prepare a PR fixing this? so we can fix this.

woutervanh, 2020-10-23:
Yes, will do.

otavio, 2020-11-05:
@woutervanh can this to be closed? I believe it is all already included in Dunfell.

closed 2021-02-01 by otavio