Freescale/meta-freescale #567: i.MX6ULL: kernel linux-imx_4.19.98.bb with CONFIG_THUMB2_KERNEL=y

fabianbuettner, 2020-11-24:
Hi,

when trying to boot this kernel, I get the following kernel panic:

> task: 88044000 task.stack: 88048000
PC is at ret_fast_syscall+0x2/0x62
LR is at SyS_brk+0xf3/0x108
pc : [<80105d22>]    lr : [<80194c07>]    psr: 60000013
sp : 88049fa8  ip : 50c53c7d  fp : 00000001
r10: 0000002d  r9 : 88048000  r8 : 80105f44
r7 : 0000002d  r6 : 00000009  r5 : 76f14819  r4 : 76f3b8d0
r3 : 88044000  r2 : ffff0001  r1 : 00000000  r0 : 01d32000
Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 50c53c7d  Table: 885d0059  DAC: 00000051
Process init (pid: 1, stack limit = 0x88048208)
Stack: (0x88049fa8 to 0x8804a000)
9fa0:                   76f3b8d0 76f14819 00000000 0000004d 00000000 00000001
9fc0: 76f3b8d0 76f14819 00000009 0000002d 00001000 00000001 00000001 00000001
9fe0: 0000002d 7e971e38 76f24155 76f24146 60000030 00000000 00000000 00000000
Code: 00004770 00000000 00000000 b6720000 (2008f8d9)
---[ end trace 096bfa5d2bdb4ef9 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b

is this a known issue? How can I fix it?

regards,
Fabian Büttner

otavio, 2020-11-24:
Em ter., 24 de nov. de 2020 às 09:53, Fabian Büttner
<notifications@github.com> escreveu:
> when trying to boot this kernel, I get the following kernel panic:
>
> task: 88044000 task.stack: 88048000
> PC is at ret_fast_syscall+0x2/0x62
> LR is at SyS_brk+0xf3/0x108
> pc : [<80105d22>] lr : [...]

We've been using Linux 5.4 for this SoC, with our customers. Any
reason to use this specific release?

-- 
Otavio Salvador                             O.S. Systems
http://www.ossystems.com.br        http://code.ossystems.com.br
Mobile: +55 (53) 9 9981-7854          Mobile: +1 (347) 903-9750

fabianbuettner, 2020-11-24:
We are currently using meta-freescale/zeus. I think that's the reason why.

otavio, 2020-11-24:
I suggest you to use Dunfell :-)

fabianbuettner, 2020-11-24:
Ok thank's for the quick response. Will report back if it works 👍

closed 2020-11-24 by otavio