Freescale/meta-freescale #601: Can not build "fsl-image-mfgtool-initramfs" image for "imx7d-pico" in Dunfell

maximdeclercq, 2020-12-21:
I'm not able to build the "fsl-image-mfgtool-initramfs" image for an "imx7d-pico" machine due to compatibility issues with "u-boot-imx-mfgtool". However the regex `(mx6|mx7|mx8)` possibly intends to match the machine `imx7d-pico`. The source code for poky only uses `re.match` instead of `re.search`, which means that the regex is only matched from the beginning of the string. Therefore `imx7d-pico` isn't matched.

The error occurs when I execute the following commands:
```shell script
$ export MACHINE="imx7d-pico"
$ export BASEBOARD="pi"
$ export DISTRO="fslc-framebuffer"
$ source setup-environment "build"
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0

Your identity is: *** <***@***.***>
If you want to change this, please re-run 'repo init' with --config-name

repo has been initialized in /work
Fetching projects: 100% (9/9), done.

Welcome to Freescale Community BSP

The Yocto Project has extensive documentation about OE including a
reference manual which can be found at:
    http://yoctoproject.org/documentation

For more information about OpenEmbedded see their website:
    http://www.openembedded.org/

You can now run 'bitbake <target>'

Common targets are:
    core-image-minimal
    meta-toolchain
    meta-toolchain-sdk
    adt-installer
    meta-ide-support

Your configuration files at build have not been touched.
$ bitbake "fsl-image-mfgtool-initramfs"
Loading cache: 100% |###########################################################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 669 entries from dependency cache.
Parsing recipes: 100% |#########################################################################################################################################################################################################################################################################################| Time: 0:00:10
Parsing of 2173 .bb files complete (381 cached, 1792 parsed). 3269 targets, 473 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing PROVIDES 'u-boot-mfgtool' (but /work/sources/meta-freescale/recipes-fsl/images/fsl-image-mfgtool-initramfs.bb DEPENDS on or otherwise requires it)
u-boot-imx-mfgtool PROVIDES u-boot-mfgtool but was skipped: incompatible with machine imx7d-pico (not in COMPATIBLE_MACHINE)
ERROR: Required build target 'fsl-image-mfgtool-initramfs' has no buildable providers.
Missing or unbuildable dependency chain was: ['fsl-image-mfgtool-initramfs', 'u-boot-mfgtool']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```

My manifest file can be found [here](https://pastebin.com/P1YahYeP).

zandrey, 2020-12-23:
You're using Community distro `fslc-framebuffer`, which is not compatible with `u-boot-imx-mfgtool`.

What you can try to do is to switch on using NXP-based distro `fsl-framebuffer` instead (note the `c` is dropped inside the name). That should effectively set a compatibility for your machine to be able to pull `u-boot-imx-mfgtool` and build _fsl-image-mfgtool-initramfs_ image.

But beware: in this case you're opting-in to use NXP version of BSP, and not Community one.

-- andrey

maximdeclercq, 2020-12-28:
Thanks a lot. Are there any downsides/differences to using the NXP distro and is this documented somewhere?

maximdeclercq, 2021-01-31:
Changing the distro to `fsl-framebuffer` still does not work, unfortunately.

Something along the dependency tree still requires `u-boot-fslc`.
```
Your build environment has been configured with:
    MACHINE = 'imx7d-pico'
    DISTRO = 'fsl-framebuffer'
    BASEBOARD = 'my-custom-board'

Building target 'my-custom-image'
Loading cache: 100% |###########################################################################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 842 entries from dependency cache.
Parsing recipes: 100% |#########################################################################################################################################################################################################################################################################################| Time: 0:00:09
Parsing of 2175 .bb files complete (530 cached, 1645 parsed). 3271 targets, 457 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies
ERROR: Nothing RPROVIDES 'u-boot-fslc' (but /work/sources/meta-freescale/recipes-fsl/packagegroups/packagegroup-fsl-mfgtool.bb RDEPENDS on or otherwise requires it)
u-boot-fslc was skipped: PREFERRED_PROVIDER_virtual/bootloader set to u-boot-imx, not u-boot-fslc
NOTE: Runtime target 'u-boot-fslc' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['u-boot-fslc']
NOTE: Runtime target 'packagegroup-fsl-mfgtool-extfs' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['packagegroup-fsl-mfgtool-extfs', 'u-boot-fslc']
ERROR: Required build target 'my-custom-image' has no buildable providers.
Missing or unbuildable dependency chain was: ['my-custom-image', 'packagegroup-fsl-mfgtool-extfs', 'u-boot-fslc']

Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
error: Command 'bitbake my-custom-image' failed.
```

I tried overriding `IMX_DEFAULT_BOOTLOADER` and `MACHINE_ESSENTIAL_EXTRA_RDEPENDS`, but `meta-freescale-3rdparty/conf/machine/imx7d-pico.conf` seems to be included after my custom conf.

```shell script
#layer.conf
...
include ${LAYERDIR}/conf/machine/${MACHINE}-extra.conf
```
machine/imx7d-pico-extra.conf
```shell script
#machine/imx7d-pico-extra.conf
...
# Use NXP based u-boot
IMX_DEFAULT_BOOTLOADER = "u-boot-imx"
MACHINE_ESSENTIAL_EXTRA_RDEPENDS = "\
    kernel-image \
    kernel-devicetree \
    u-boot-imx \
"
```

This is the output of `bitbake -e`
```shell script
...
# $IMX_DEFAULT_BOOTLOADER [3 operations]
#   set /work/sources/meta-custom/conf/machine/imx7d-pico-extra.conf:13
#     "u-boot-imx"
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:9
#     "u-boot-fslc"
#   override[mx8]:set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:10
#     "u-boot-imx"
# pre-expansion value:
#   "u-boot-fslc"
IMX_DEFAULT_BOOTLOADER="u-boot-fslc"
...
# $MACHINE_ESSENTIAL_EXTRA_RDEPENDS [4 operations]
#   set /work/sources/meta-custom/conf/machine/imx7d-pico-extra.conf:18
#     "    kernel-image     kernel-devicetree     u-boot-imx "
#   append /work/sources/meta-freescale-3rdparty/conf/machine/imx7d-pico.conf:41
#     "     kernel-image     kernel-devicetree     u-boot-fslc "
#   set /work/sources/core/meta/conf/documentation.conf:277
#     [doc] "A list of required machine-specific packages to install as part of the image being built. Because this is a 'machine essential' variable, the list of packages are essential for the machine to boot."
#   set? /work/sources/core/meta/conf/bitbake.conf:823
#     ""
# pre-expansion value:
#   "    kernel-image     kernel-devicetree     u-boot-imx       kernel-image     kernel-devicetree     u-boot-fslc "
MACHINE_ESSENTIAL_EXTRA_RDEPENDS="    kernel-image     kernel-devicetree     u-boot-imx       kernel-image     kernel-devicetree     u-boot-fslc "
...
#
# $PREFERRED_PROVIDER_nativesdk-u-boot-mkimage
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:29
#     [_defaultval] "nativesdk-${IMX_DEFAULT_BOOTLOADER}-tools"
PREFERRED_PROVIDER_nativesdk-u-boot-mkimage="nativesdk-u-boot-fslc-tools"
#
# $PREFERRED_PROVIDER_nativesdk-u-boot-tools
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:27
#     [_defaultval] "nativesdk-${IMX_DEFAULT_BOOTLOADER}-tools"
PREFERRED_PROVIDER_nativesdk-u-boot-tools="nativesdk-u-boot-fslc-tools"
...
#
# $PREFERRED_PROVIDER_u-boot-mkimage-native
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:28
#     [_defaultval] "${IMX_DEFAULT_BOOTLOADER}-tools-native"
PREFERRED_PROVIDER_u-boot-mkimage-native="u-boot-fslc-tools-native"
#
# $PREFERRED_PROVIDER_u-boot-mxsboot-native
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:32
#     [_defaultval] "u-boot-fslc-mxsboot-native"
PREFERRED_PROVIDER_u-boot-mxsboot-native="u-boot-fslc-mxsboot-native"
#
# $PREFERRED_PROVIDER_u-boot-tools-native
#   set /work/sources/meta-freescale/conf/machine/include/imx-base.inc:26
#     [_defaultval] "${IMX_DEFAULT_BOOTLOADER}-tools-native"
PREFERRED_PROVIDER_u-boot-tools-native="u-boot-fslc-tools-native"
...
```

otavio, 2021-01-31:
I'd rather fix the U-Boot FSLC to support the mfgtool. The use-nxp-bsp is not necessarily referring to all NXP GA specific components but the Vivante GPU drivers and Linux fork.

maximdeclercq, 2021-01-31:
Thanks for the clarification, I'll see if I'm able to fix the `u-boot-fslc` to support the mfgtool.

maximdeclercq, 2021-01-31:
https://github.com/Freescale/meta-freescale/pull/627 allows the `mfgtool` to build with the `fslc` distros. I'll check if this produces a  flashable image.

otavio, 2021-02-01:
Fixed ... closing.

closed 2021-02-01 by otavio