Freescale/meta-freescale #611: [dunfell] xf86-video-armada fetch failure

sakoman, 2021-01-12:
WARNING: xf86-video-armada-2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0 do_fetch: Failed to fetch URL git://git.arm.linux.org.uk/cgit/xf86-video-armada.git;branch=unstable-devel;protocol=http;name=armada, attempting MIRRORS if available
ERROR: xf86-video-armada-2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; unset _PYTHON_SYSCONFIGDATA_NAME; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1473"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/steve/builds/rt-yocto-setup-fsl/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/steve/builds/rt-yocto-setup-fsl/sources/poky/scripts:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot-native/usr/bin/arm-reach-linux-gnueabi:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot/usr/bin/crossscripts:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot-native/usr/sbin:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot-native/usr/bin:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot-native/sbin:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/work/armv7at2hf-neon-reach-linux-gnueabi/xf86-video-armada/2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0/recipe-sysroot-native/bin:/home/steve/builds/rt-yocto-setup-fsl/sources/poky/bitbake/bin:/home/steve/builds/rt-yocto-setup-fsl/build/tmp/hosttools"; export HOME="/home/steve"; LANG=C git -c core.fsyncobjectfiles=0 clone --bare --mirror "http://git.arm.linux.org.uk/cgit/xf86-video-armada.git" /home/steve/yocto-files/dunfell/downloads/git2/git.arm.linux.org.uk.cgit.xf86-video-armada.git --progress failed with exit code 128, output:
Cloning into bare repository '/home/steve/yocto-files/dunfell/downloads/git2/git.arm.linux.org.uk.cgit.xf86-video-armada.git'...
fatal: repository 'http://git.arm.linux.org.uk/cgit/xf86-video-armada.git/' not found

ERROR: xf86-video-armada-2_0.1+gitAUTOINC+78e7116a5b_8478eef32f-r0 do_fetch: Fetcher failure for URL: 'git://git.arm.linux.org.uk/cgit/xf86-video-armada.git;branch=unstable-devel;protocol=http;name=armada'. Unable to fetch URL from any source.

otavio, 2021-01-13:
Have you tried this on master?

sakoman, 2021-01-13:
No, I haven't.  I'm preparing the 3.1.5 release of dunfell LTS oe-core, and one of my tests is to do a clean build with some popular layers.  That's where I discovered the issue.

I see that you've done a related patch for master:

https://github.com/Freescale/meta-freescale/commit/2b18a6b0be178b0207c86fa049271cdd4c815981#diff-65124f1ece90a276110cb5089ed42994170493a4b9102134b6572a5c02fbe801

This probably should be backported to dunfell too.

Thanks!

otavio, 2021-01-13:
Applied.

closed 2021-01-13 by otavio