Freescale/meta-freescale #622: Integrate NXP Release 5.4.70_2.3.0

MaxKrummenacher, 2021-01-29:
@zandrey already worked on this in his fork
https://github.com/zandrey/meta-freescale/tree/upgrade-to-rel_imx_5.4.70_2.3.0

I added further patches on top of that here:
https://github.com/MaxKrummenacher/meta-freescale/commits/upgrade-to-rel_imx_5.4.70_2.3.0
I successfully built it for a 64bit and a 32bit machine. I have yet to try it out on target, so IMHO at least my work is not ready for merge.

Just wanted everyone to know so we do not duplicate work.

otavio, 2021-01-29:
Awesome. @zandrey can you prepare a PR?

zandrey, 2021-01-29:
@MaxKrummenacher Thanks for the update and follow-up on my branch!

@otavio I guess it would be fair to ask Max if he would make a PR as he took my (relatively small) base and went through the rest of packages.

@MaxKrummenacher would you be willing to submit a PR of your branch, or should I take over all commits from your fork and prepare a PR myself?

-- andrey

MaxKrummenacher, 2021-01-29:
Hi

@otavio  @zandrey Ok, 

Sure no problem, I do it. I guess from your comment you are happy with your patches as is. I will first test two images on the target and rebase on current master.

zandrey, 2021-01-29:
> Sure no problem, I do it.

Thanks a lot!

> I guess from your comment you are happy with your patches as is.

I believe I had the "base" part covered, and so far `core-image-base` was operable for me on `imx8mm-lpddr4-evk` machine. 

> I will first test two images on the target and rebase on current master.

I was planning to do the same on my end, but I can do this also with the PR patch.

-- andrey

MaxKrummenacher, 2021-02-02:
Upgrade merged, closing this issue.

closed 2021-02-02 by MaxKrummenacher